PX4/PX4-Autopilot #21687: Primary EKF changed 1 (unhealthy) -> 0 on PX4 v1.13.2 causing crash of the drone

Katawann, 2023-06-06:
## Describe the bug
We were flying with our drone above vineyards when suddenly the motors of the drone stop and the drone fall down in the vineyard... From the log and the 3D flight, I can see a big jump in altitude positioning and the EKF resetting but I will need some experts to help me to understand the root cause...

## Log Files and Screenshots
Log of the file can be downloaded here: https://logs.px4.io/plot_app?log=36440ff4-6aaf-4dd7-9a29-9fb9bd9ad3d8

[image]
[image]
[image]

## Drone:
- Hexacopter with PX4 v1.13.2 and Pixhawk 4
- Primary GPS is CUAV ublox F9P
- Secondary GPS is Holybro M8N

bresch, 2023-06-07:
Hi, sorry to hear about the crash. The EKF instance switch seems to be a consequence of the issue that led to the crash and not the cause of it.
As you can see in the plot below, all motors suddenly stop for about 1s (current goes down to 1A, gyro signals are flat, step in the mag data due to the sudden current drop). What's really strange is that both the baro and both GNSS altitudes are stepping by 8m at the exact same moment. It looks like a couple of seconds are missing, but not due to logging dropouts: we don't see the drone falling in the log, it's like it went down in 0 seconds

@dagar  did it maybe reboot in air and the logger just appended the data (not leaving a gap)?

[image]

Igor-Misic, 2023-06-07:
If see correctly this was 8 meters fall, so everything happened under 1 second.
My guess is that ESCs get restarted in the flight or they lost power or connection for 1 second. That would explain the current reduction to above 2 amps and free fall.
Now, we are missing more log dots for freefall since it is logged at a 200 ms rate, so we should get at least 4 dots for fall. 
But I would remove 2 dots because we are looking at filtered data.

closed 2023-06-12 by bresch