PX4/PX4-Autopilot #21702: Hold/Loiter continuing to next WP before Holding

ryanjAA, 2023-06-08:
1.14 beta (exact commit in log below).

When in mission mode and hitting hold, instead of stopping and loitering, it continues to the next WP. Then stops and loiters. I did test this on a real aircraft and didn't see this issue but I'll have to go back and see.

Flight Log:
https://review.px4.io/plot_app?log=2dffea97-efce-4a7c-9dc3-16ad4a32667f

Mission Plan (just for reference):
[Hold issue.plan.zip](https://github.com/PX4/PX4-Autopilot/files/11691414/Hold.issue.plan.zip)

If this is intentional (doubt it) then should be clear somewhere and QGC should not say will hold at current location, etc, etc 

Should also block 1.14 if unintentional. @dagar

sfuhrer, 2023-06-29:
@ryanjAA thanks for the report! Took me a while how to reproduce it as I never use the direct flight mode selection in QGC and instead always click on the "Hold" button. Seems like we have different behaviors between them, and in https://github.com/PX4/PX4-Autopilot/pull/21775 I'm addressing it. Will port to 1.14 once in main.

mrpollo, 2023-07-04:
Closed by https://github.com/PX4/PX4-Autopilot/pull/21775

closed 2023-07-04 by mrpollo