PX4/PX4-Autopilot #21715: [Bug]: VectorNav INS driver causing airspeed sensor to report a fault

lashVN, 2023-06-12:
### Describe the bug

Using the VectorNav driver in EKF2 disable mode, the airspeed sensor reports a failure after a short period of time since startup. 

### To Reproduce

Configure PX4 to use VectorNav driver on correct branch with a fixed-wing aircraft
Disable EKF2 (instead opting for VectorNav INS)
Takeoff, fly for a few minutes
Observe airspeed sensor ceasing to report data

### Expected behavior

Airspeed sensor continues to report data throughout flight

### Flight Log

```txt
To be attached
```

### Software Version

```txt
pr-vectornav_ins
```

### Flight controller

Pixhawk V6C

### Vehicle type

Hybrid VTOL

sfuhrer, 2023-07-04:
The main check of the airspeed validator is comparing the airspeed to ground speed and an estimated wind speed. In order to estimate the wind speed we need to fuse ground speed and vehicle heading as well, which normally come from the main estimator on board. The test ratios are only there as an extra layer of protection against estimator instabilities, and it can well be that we can get rid these checks and instead use more generic estimator flags that indicate if certain states are good to be used or not.

lashVN, 2023-07-05:
> it can well be that we can get rid these checks and instead use more generic estimator flags that indicate if certain states are good to be used or not

Great, is this something that needs to be investigated, or just implemented? Either way, what would it take to prioritize this fix?

dagar, 2023-07-06:
In the meantime here's a way to have more direct control skipping the airspeed selector. https://github.com/PX4/PX4-Autopilot/pull/21813