PX4/PX4-Autopilot #21721: [Bug]: IR-Lock Precision Landing LTEST_ Parameters Missing

Ivan-ValDev, 2023-06-14:
### Describe the bug

I prepared a drone setup using a Cube Black, IR-Lock and Lightware SF11/C distance sensor to perform precision landing.
After following the steps from [PX4 precision landing](https://docs.px4.io/main/en/advanced_features/precland.html) I saw that all the LTEST_ related parameters were not showing up. As a result the drone that I prepared has not performed precision landing so far.

### To Reproduce

### Check if the LTEST_ related parameters are available in QGC
1. Open QGC
2. Click on QGC icon on the top left corner
3. Select Vehicle Setup
4. Click on Parameters tab from the left column
5. on the search bar look for SENS_EN_IRLOCK and for LTEST
6. If none of those shows then it is possible that Landing Target Estimator is not running.

### Expected behavior

- The LTEST_ parameters should have showed up in QGC
- The drone should have at least tried to the precision landing sequence giving the situation:
   - Drone is flying at least 10 meters above an IR beacon
   - The user triggers precision landing from QGC's main window or MAVLink console

### Flight Log

```txt
There is no log for this issue
```

### Software Version

```txt
1.13.3 px4 fmu-v3
```

### Flight controller

Cube Black

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

- IR-Lock sensor connected over an I2C bus
  - enable this parameter in QGC -> SENS_EN_IRLOCK = 1 (ENABLED) 
- Lightware SF11/C distance sensor over I2C bus
  - enable this parameter in QGC -> SENS_EN_SF1XX = 4 (SF11/c with I2C)

scheysej, 2023-06-16:
Landing_Target_Estimator is a module that needs to be started.

You can do this manually in QGC:
- Click in the top left corner and hit Analyze Tools.
- Navigate down to MAVLink Console
- Type `landing_target_estimator start`
- The parameters should be showing.

If you wish it to start on boot:
- Remove the microSD from the CubeBlack and plug it into your computer.
- Create a folder called `etc` and navigate into it
- Create a text filed called `extras` and open it
- Inside of the text file write `landing_target_estimator start` and save the file
- Eject the microSD from your computer and plug it back into the cube.
- Reconnect to QGC and you should see your parameters.

Ivan-ValDev, 2023-06-16:
@scheysej thanks a lot. Actually I came to a solution similar to that one.

Now the parameters show up but the test drone that I'm using acts erratically instead of trying to center over the beacon. This might be for a new issue but I'll leave it open for a while to see if there is any advice on this

Ivan-ValDev, 2023-06-16:
Also this issue was mentioned in the PX4 community call from June 14, 2023 [link](https://discuss.px4.io/t/px4-community-q-a-june-14-2023/32616)

DavidPaddy97, 2023-07-04:
We fixed this by adding the following to /ROMFS/px4fmu_common/init.d/rcS at line 538 which eliminates the need to edit the SD card of every drone you fly:

```
	#
	# Start the Landing Target Estimator for precision land
	#
	if param compare -s SENS_EN_IRLOCK 1
	then
		landing_target_estimator start
	fi
```

[image]

wolfie93, 2025-03-13:
Hello!

After following steps suggested by @scheysej, I was able to get LTEST parameter, whereas SENS_EN_IRLOCK didn't show up. I am using Pixhawk 6C flight controller with PX4 firmware version 1.16.0alpha (PX4 Pro - Developer Build(master)). Is SENS_EN_IRLOCK parameter restricted to certain flight controllers?

bresch, 2025-03-13:
@wolfie93 The IR lock driver is not compiled by default on the 6C board, you need to enable it using the [menuconfig tool](https://docs.px4.io/main/en/hardware/porting_guide_config.html#px4-menuconfig-setup).
So run `make px4_fmu-v6c_default boardguiconfig`, enable `irlock` in the drivers section, save and re-build/flash your autopilot (`make px4_fmu-v6c_default upload`).

closed 2025-03-13 by bresch

wolfie93, 2025-05-05:
Hello @bresch .

I tried using your method. But it looks like the drone was unable to find the beacon. Can you please if the drone entered precision land mode or not? Did it IR LOCK? And also, the drone crashed few seconds after switching to precision landing mode. FYI, precision landing worked perfectly at same height on same drone using Arducopter and Mission Planner.

You can find the log here(https://review.px4.io/plot_app?log=72999c48-ea56-4f94-9acd-2d69755edd98).