PX4/PX4-Autopilot #21730: [Bug]: Not able to control all the drone in Multi Drone simulation.

Apurv354, 2023-06-16:
### Describe the bug

I am trying to create a multi-drone simulation following this document - https://docs.px4.io/main/en/ros/ros2_multi_vehicle.html.
I am trying to spawn 4 drones and send commands through ROS2. One of the drones is not responding. I have tried changing the namespace but it doesn't work. The vehicle numbers are 2,3,4,5 and the problem happens in the 5th vehicle.

### To Reproduce

Run - sitl_multiple_run.sh with 4 px4_instance. 
Run - Write a Publisher similar to this - https://github.com/PX4/px4_ros_com/tree/main/src/examples/offboard_py for 4 drones. 

### Expected behavior

Target System 5 will not respond to the commands from the publisher. 

### Flight Log

```txt
https://logs.px4.io/plot_app?log=c7e2d06c-7a75-4ac2-a597-6e898f36bcb6
```

### Software Version

```txt
v1.13.3
```

### Flight controller

Pixhawk

### Vehicle type

Multicopter

beniaminopozzan, 2023-06-16:
Hi @Apurv354 

I see a huge amount of preflight fails in the log you should fix those before looking into ROS 2
I see the from the log the drone moving even if all motors are stopped, there is something wrong with the simulation here

Apurv354, 2023-06-18:
> 

I am using this [script](https://github.com/PX4/PX4-Autopilot/blob/main/Tools/simulation/gazebo-classic/sitl_multiple_run.sh) from PX4 to launch 4 drones in the simulation. Pre-flight fails are happening only in one drone I will try to fix them but all the other drones are working fine.

closed 2024-01-09 by Apurv354