PX4/PX4-Autopilot #21734: [Bug]: 6x CANFD port and Ethernet can't use at the same time.

DaShen96, 2023-06-19:
### Describe the bug

There’s a problem bothering me. Why is the Ethernet device missing after configuring the CAN port?

```
nsh> ifconfig
can0	Link encap:UNSPEC at DOWN
	inet addr:0.0.0.0 DRaddr:0.0.0.0 Mask:0.0.0.0

can1	Link encap:UNSPEC at DOWN
	inet addr:0.0.0.0 DRaddr:0.0.0.0 Mask:0.0.0.0

nsh> 
```

After disabling the CAN port, the Ethernet port can work normally.

```
nsh> ifconfig
eth0	Link encap:Ethernet HWaddr 96:8b:9c:ce:b9:21 at DOWN
	inet addr:192.168.0.54 DRaddr:192.168.0.1 Mask:255.255.255.0
```

### To Reproduce

1. enable FDCAN on menuconfig , can't see the eth0 port use ifconfig command.
2. disable FDCAN on menuconfig , the eth0 port work normally.
3. Ethernet configuration is like that:
```
nsh> netman show
DEVICE=eth0
BOOTPROTO=static
NETMASK=255.255.255.0
IPADDR=192.168.0.54
ROUTER=192.168.0.1
DNS=192.168.0.1
```

### Expected behavior

I need to use both CAN and Ethernet at the same time.

### Flight Log

```txt
none
```

### Software Version

```txt
v1.14.0
```

### Flight controller

pixhawk 6x

### Vehicle type

None

cody-czn, 2024-04-18:
I have observed the same issue

DaShen96, 2024-06-21:
> I have observed the same issue

Did you solve this issue?

cody-czn, 2024-06-26:
I found the reason for the ethernet interface disappearing.

The issue is that the ethernet driver does not get initialised in Nuttx when CANFD  is enabled.

See below requirement for multiple interfaces.

 `arch/arm/src/stm32h7/stm32_ethernet.c`.
```c

/****************************************************************************
 * Function: arm_netinitialize
 *
 * Description:
 *   This is the "standard" network initialization logic called from the
 *   low-level initialization logic in arm_initialize.c. If STM32H7_NETHERNET
 *   greater than one, then board specific logic will have to supply a
 *   version of arm_netinitialize() that calls stm32_ethinitialize() with
 *   the appropriate interface number.
 *
 * Parameters:
 *   None.
 *
 * Returned Value:
 *   None.
 *
 * Assumptions:
 *
 ****************************************************************************/

```

DaShen96, 2024-06-26:
> I found the reason for the ethernet interface disappearing.
>
> The issue is that the ethernet driver does not get initialised in Nuttx when CANFD is enabled.
>
> See below requirement for multiple [...]

Thank for your reply! I found that too. I have modified the NuttX driver, and now it works properly.

pps90, 2024-11-03:
Can you explain how to fix this bug

DaShen96, 2024-11-04:
> Can you explain how to fix this bug

Perhaps this text can help you.
1、PX4-Autopilot/boards/px4/fmu-v6x/nuttx-config/nsh/defconfig
Add these configuration:
`CONFIG_NETDEV_LATEINIT=y
CONFIG_NETDEV_IFINDEX=y
CONFIG_NETDEV_CAN_BITRATE_IOCTL=y
CONFIG_NET_CAN=y
CONFIG_NET_CAN_RAW_TX_DEADLINE=y
CONFIG_NET_CAN_SOCK_OPTS=y
CONFIG_NET_CAN_NOTIFIER=y
CONFIG_NET_TIMESTAMP=y
CONFIG_STM32H7_FDCAN1=y
CONFIG_STM32H7_FDCAN2=y
CONFIG_CANUTILS_CANDUMP=y
CONFIG_CANUTILS_CANSEND=y
CONFIG_FDCAN1_ARBI_BITRATE=500000
CONFIG_FDCAN1_DATA_BITRATE=2000000
CONFIG_FDCAN2_ARBI_BITRATE=500000
CONFIG_FDCAN2_DATA_BITRATE=2000000`

2、PX4-Autopilot/boards/px4/fmu-v6x/src/init.c
add these header file:
`#include "stm32_fdcan_sock.h"
#include "stm32_ethernet.h"`
Add the following code to the tail of the __EXPORT int board_mapp_initialize (uintptr_t arg) function:
`#if STM32H7_NETHERNET > 0 && defined(CONFIG_STM32H7_ETHMAC)
stm32_ethinitialize(0);
#endif
#ifdef CONFIG_STM32H7_FDCAN1
stm32_fdcansockinitialize(0);
#endif
#ifdef CONFIG_STM32H7_FDCAN2
stm32_fdcansockinitialize(1);
#endif`

3.maybe you should modify the parameters following this hack:
#https://github.com/julianoes/PX4-NuttX/commit/b39310baea123ee27333e1743a80bbc7bd5c74e7

pps90, 2024-11-04:
ok! thank you!!