PX4/PX4-Autopilot #21741: [Bug]: Low real time factor for Gazebo simulation of multiple boats

ShiauweiZhao, 2023-06-20:
### Describe the bug

cpu i9-10900k 
gpu rtx3090 
ram 64G  but real time factor is 0.07
[image]

### To Reproduce

Modify Tools/simulation/gazebo-classic/sitl_multiple_run.sh   Add boat type  and run 

Tools/simulation/gazebo-classic/sitl_multiple_run.sh -m boat -n 2 -w boat

### Expected behavior

real time factor is 1

### Flight Log

```txt
no log
```

### Software Version

```txt
branch release/1.14  commit id d9585bf3d3aa59a126fb061c644cdd4960442d3c
```

### Flight controller

sitl

### Vehicle type

Boat

junwoo091400, 2023-06-29:
@khanasif786 Do you perhaps have clue on this? Since you recently submitted PR on Gazebo!

khanasif786, 2023-07-02:
@junwoo091400 . I will tell you soon after testing.

khanasif786, 2023-07-03:
@junwoo091400 @ShiauweiZhao In my case the RTF is 0.1 average
CPU i3 2120 (Oldest in the world)
GPU GT1030 (Oldest in the world)
[image]

khanasif786, 2023-07-03:
I dont think boat is supported That's what it tell if boat is not added in supported vehicles manually. But i will test this again in another system soon.

khanasif786, 2023-07-03:
BTW I think @ShiauweiZhao you can solve this by playing with physics parameteres of gazebo.