PX4/PX4-Autopilot #21748: [Bug][EKF2]: Height estimate error

w2016561536, 2023-06-21:
### Describe the bug

The height estimate error occured
My drone is equipped with lidar, but not GPS. The height data (Local position Z) was "reset" while flying. But the distance sensor and barometer worked fine. 
In some ways, it was my operation fault (perhaps no one would switch take off to land). But i wish it can be fixed.

### To Reproduce

Take off the flight at position mode. Stay at Position mode. Then switch to take off mode again, and then directly switch to land mode. Then, the drone fly up to the ceiling.

### Expected behavior

The height data should not jump to the original data.

### Screenshot / Media

[image]

### Flight Log

```txt
https://review.px4.io/plot_app?log=41c542ee-616b-47f4-904a-9d4e0c227150
```

### Software Version

```txt
1.13.3, but compiled by myself(update to commit 1c8ab2a)
```

### Flight controller

Pixhawk 2.4.8

### Vehicle type

Multicopter

julianoes, 2023-10-19:
@bresch is this still valid?

bresch, 2023-10-26:
The jump was caused by the old "range aid" logic (resets when switching between baro and range fusion modes). This is done differently now, please try again with 1.14 and re-open an issue if the problem persists.
[image]

closed 2023-10-26 by bresch