PX4/PX4-Autopilot #21750: Multiple airspeed sensors

DenisPGH, 2023-06-22:
### Describe problem solved by the proposed feature

I read some previous pull requests since 4 years and I understand, there is no implementation of two or more airspeed sensors because of ability for calibration, ekf precision etc.. I know the importance of good airspeed sensor or two is very big, therefor we want to implement in our project two airspeed sensors, working simultaneously, so I think the errors possibility or crash will be very minimize. I would like to ask, Is there some solution already or it must be implemented? 

### Describe your preferred solution

Build implementation of second airspeed sensor and calibration possibility for two airspeed sensors. 

### Describe possible alternatives

I am not sure. If someone has, please.

DenisPGH, 2023-07-26:
I did this in this way below. I will be very happy, if got some review:
1. In UAVCAN added splitting by address of the differential pressure sensors, first(or all with same address) send it data to differential_pressure topic, the second sent its data to differential_pressure_second topic, which I created.
2. In sensor.cpp added one more function with airspeed_second topic.
3.  Added calibration logic, messages in mavlink and button in QGC(so the second sensor can be also calibrated).
4. Added new fields in VFR_HUD message for transfer the data PX4 <-> QGC. The Information from both sensors can be seen in the App(Vehicle).
5. In airspeed_selector store the data not in airspeed_validated, but in airspeed_validated_new topic(clone of airspeed_validated, created from me).
6. Created new module airspeed_fusion. Where the both airspeed_validated topics from both sensors are subscribed and according of "airspeed_sensor_measurement_valid" variable of each topic, it has tree ways for publishing to airspeed_validated:
- both are true-> made basic ekf for fusion the sensors' data
- only first is true -> get data only from the first.
- only second is true -> get data only from the second sensor.

DenisPGH, 2023-07-26:
Test video:
Air Spd : combined value from the ekf.
Air Spd 2: Airspeed from second sensor
Air Spd 3: Airspeed from first sensor
https://github.com/PX4/PX4-Autopilot/assets/103575909/d3bbc7ce-98fd-4536-868f-4c8c6fc1464b

junwoo091400, 2023-09-09:
Have you checked out this documentation?

https://discuss.px4.io/t/jmavsim-not-found-during-installation-ninja-error-unknown-target-jmavsim-did-you-mean-jsbsim/33869/2

julianoes, 2023-10-19:
@DenisPGH any response to @junwoo091400's doc link?

DenisPGH, 2023-10-20:
@julianoes, @junwoo091400   Hello, I read the doc link, but our idea is different. We want two complete separately airspeed sensors, which work simultaneously, and both measured data is collected and used in the main calculation.

RoyCRO, 2023-10-24:
@DenisPGH Hello, I am interested to know more about your idea. I am currently trying to connect 2 airspeed sensors 
(MS4525DO) to Pixhawk using an I2C splitter, but only one airspeed is shown on QGC and I would like to be able to see the airspeed from both sensors at the same time. Could you share more details about how you managed it? Thanks!

tnedev, 2023-10-24:
I work together with @DenisPGH. 

We wanted to use two sensors (or possibly more) because frequently those sensors cause issues and it's a single point of failure. This is especially critical for bigger UAVs and we don't want to rely on a single sensor. 

We've added two UAV CAN sensors and developed a small EKF to handle the two sensors data and feed back the data to the main EKF.  It works well and it was tested with a few successful flights. 

The plan was to add this as contribution to PX4, but because it involves a lot of changes in PX4, MAVLink, QGC and more for now it will be too much hassle and probably not aligned with PX4 goals. Just the PX4 changes affect around 60 files.

jessie-vandam, 2024-08-15:
@DenisPGH @tnedev I have just implemented [https://github.com/PX4/PX4-Autopilot/pull/22785](url) and indeed two airspeed topics are being published now, with two connected airspeed sensors. The airspeed calibration is still only possible for one of the sensors though. Can you give a bit more detail about the changes in the airspeed calibration part, or are there also to many file changes involved with that?

bermuda-pilot, 2026-10-13:
Hello, Any update on this Dual airspeed sensor concept,
I want to try 1 I2C and 1 CAN airspeed sensor,
would require external EKF calculation to measure both values and feed selected value?