PX4/PX4-Autopilot #21770: [Bug]: RTL_TYPE = 0 not working on 1.14beta

jstrebel, 2023-06-28:
### Describe the bug

ERROR Mission rejected: Landing waypoint/pattern required.
RTL_TYPE 0 has no effect.
With 1.13.3, same Mission all OK

SW v1.14.0 (beta) ([18898f18](https://github.com/PX4/Firmware/commit/18898f1876ec1be9f86d89aedc56b4918c983afa))
HW: pixhawk4
QGC:4.2.6
log:https://review.px4.io/plot_app?log=a2bd448c-ad95-4b88-a6cf-bafb72a37e3a

### To Reproduce

RTL_TYPE 0 has no effect.
With 1.13.3, same Mission all OK

### Expected behavior

RTL_TYPE = 0 works in 1.13.3

### Flight Log

```txt
https://review.px4.io/plot_app?log=a2bd448c-ad95-4b88-a6cf-bafb72a37e3a
```

### Software Version

```txt
1.14beta and QGC 4.2.6
```

### Flight controller

pixhawk4

### Vehicle type

Fixed Wing

junwoo091400, 2023-06-29:
Thanks for creating this report! However, have you checked Silvan's comment here? -> https://github.com/PX4/PX4-Autopilot/issues/21358#issuecomment-1612653896

[image]

As he said, it seems that `MIS_TKO_LAND_REQ` parameter (added between v1.13.3 and v1.14 beta), which you have set to `2` is the cause of this error. This is currently the default value for VTOL (mandatory to have landing pattern).

For a quick fix, you can override this by setting `MIS_TKO_LAND_REQ` parameter to `0`, if you wish to not have takeoff, nor landing pattern. Hope this helps!

junwoo091400, 2023-07-04:
@jstrebel Please check Silvan's comment: You can customize requiring takeoff/landings in missions through MIS_TKO_LAND_REQ, see also https://docs.px4.io/main/en/flight_modes/mission.html#mission-feasibility-checks.

This should resolve your issue!

closed 2023-07-04 by junwoo091400