PX4/PX4-Autopilot #21771: [Bug]: No Simulation in Hardware on V1.14beta2

aminballoon, 2023-06-29:
### Describe the bug

It doesn't have Simulation in Hardware on V1.14beta2. No fakeGPS fakeLocalposition
I want to use UXRCE to interfaces with ros2 and POC it SIH system . 

### To Reproduce

[image]

### Expected behavior

Can use SIH on PX4-Autopilot v1.14.0-beta2 

### Flight Log

```txt
-
```

### Software Version

```txt
PX4-Autopilot v1.14.0-beta2 Firmware
```

### Flight controller

CUAV v6x  

### Vehicle type

Other

junwoo091400, 2023-06-29:
Thanks for the report. However, what is the bug exactly? That it's having preflight failures? Or that fake GPS is not working (how did you verify that)?

aminballoon, 2023-06-29:
@junwoo091400 fake_gps is not found in nuttx .
[image]
Does you see in this picture there are didn't have folder examples inside **src** after build with command `make px4_fmu-v6x_default`

aminballoon, 2024-01-17:
@junwoo091400 Hi, Do you have any update for this issue with CUAV-v6x ?

closed 2024-04-22 by aminballoon

junwoo091400, 2024-04-22:
Ji @aminballoon, did you resolve this issue?

aminballoon, 2024-04-26:
@junwoo091400 Yes, I have add SIH and fakeGPS module in .px4board file. Does anyone want some example ?