PX4/PX4-Autopilot #21774: [Bug]: Runway Takeoff / Landing bugs and problems.

tubeme, 2023-06-29:
### Describe the bug

Hello guys, long time no see. Glad that the progress of PX4 is tremendous, but there are some unexplored corners  like the Runway Takeoff / Landing, that need some attention.

Basically the autonomous runway takeoff and landing in a mission does not work correctly and it works half way. Only the take off is working but with some bugs. (We haven't tested the the auto runway takeoff mode with a switch, this is only the mission part).

1. During takeoff phase, there is some incorrect YAW going to the left. The first time we tested to takeoff it just turned left i switched back to stabilized but it crashed on the ground because of the speed. See LOG 1. This time we preset RWTO_HDG = 1 (1: position control along runway direction).
2. On a second day we made the same thing but with RWTO_HDG = 0 (0: airframe heading when takeoff is initiated), but with the same result. This time I managed to switch off the mission and save the craft. See LOG 2.
3. Third time we engaged the mission I used the YAW stick to just manually push the YAW a little bit in order to make the correct heading and it took off without a problem, but still the left YAW was persistent. (I pushed it a little more to the right, but still the take off was successful.) See LOG 3.
4. You can see in the screenshot how we setup the mission. There is a problem though, even if the aircraft is setup to use runway takeoff and landing its not correct visually in the QGC. In QGC it looks like we are throwing the aircraft from hand, and the initial climb is not correctly shown. 
5. There were no FLAPS lowered during the runway takeoff, I suppose they are not engaged during the landing as well. We haven't tested the landing gear automation but most likely it is the same as the FLAPS.
6. The autonomous runway landing did NOT WORK. It just went past the landing pattern, went over the home spot over our heads and then started loitering at the end of the mission. I just switched off the mission and landed manually.

Those are the problems so far. 

We can work on the problems with our software team, but we will appreciate some help from the community, to identify first what is going on and to see where we have to change the code in order to correct the problems.

At a later stage I will make a nice visualizations for the Users Manual, because it mostly lacking Runway Takeoff and Landing information.

Thank you very much in advance.

Vasil

### To Reproduce

Nothing special to reproduce it. It happens every time we program a mission with autonomous runway takoff / landing

### Expected behavior

It is expected to have a fully autonomous takeoff/landing without human intervention.

It is supposed the runway takeoff/landing to be correctly visualized in the QGC. (The innitial acceleration on the ground, takoff and then climb to the takoff altitude)  

It is supposed to use FLAPS during takoff and landing. 

It is supposed to close the retracts after takoff and release them on landing.

### Screenshot / Media

[image]

### Flight Log

```txt
1. LOG - runway takeoff left YAW - RWTO_HDG = 1
https://review.px4.io/plot_app?log=2e91edf9-9731-4d62-8aab-82e320c5f424

2. LOG - runway takeoff left YAW - RWTO_HDG = 0
https://logs.px4.io/plot_app?log=3d4378a3-520b-40ed-bcee-1f41458ff9f0

3. LOG - successful take off and mission but no landing.
https://logs.px4.io/plot_app?log=127ef61b-b5b7-438e-ba30-79b4b58f3196
```

### Software Version

```txt
We are using the latest masters most of the time. You can see which master we used from the logs.
```

### Flight controller

Its is FMUv5 - CUAV V5+ with CAN power brick, airspeed and all the sensors.

### Vehicle type

Fixed Wing

### How are the different components wired up (including port information)

It is wired correctly. This bugs are not from connections.

We have CAN power brick, Airspeed is I2C Sensirion SDP3x.

tstastny, 2023-07-09:
@tubeme  - glad some others are using fixed-wing runway modality! Happy to have feedback on things we can improve or if anything has broken. I took a look at your logs, and believe you may have missed some documentation, however - please see below.

> During takeoff phase, there is some incorrect YAW going to the left.

I took a look at the log, and you don't have wheel control enabled, i.e. `FW_W_EN`. This means that your wheel is getting a constant input of **zero** (open loop, no controller running). So when you held full yaw stick, it was only the force from the vertical tail (from rudder deflection) helping you stay straight, not wheel nudging. See in the plots below that when you switch to takeoff mode, the wheel input is zero for the whole ground roll:
[image]

> There were no FLAPS lowered during the runway takeoff

You need to set the flaps scale for takeoff `FW_FLAPS_TO_SCL`. It was set to **zero** in your flights.

> I suppose they are not engaged during the landing as well

Additionally need to set this yourself via `FW_FLAPS_LND_SCL` -- it looks like that you did have set to 1.0 in your flights.

> In QGC it looks like we are throwing the aircraft from hand, and the initial climb is not correctly shown.

I agree the visualization of takeoff could definitely be enhanced, if you are open to helping with this, would be awesome! For now, it is documented that the line from start position to takeoff waypoint is what the vehicle will track on ground and in air indefinitely until reaching the clearance altitude. With the current reliance on waypoints to command this, the current visualization was the minimal solution.

I'd encourage you to please read through the documentation on how to use (and tune)

**runway takeoff** here: https://docs.px4.io/main/en/flight_modes/takeoff.html#runway-takeoff
**runway landing** here: https://docs.px4.io/main/en/flight_modes/mission.html#fw-mission-landing

There is quite some detailed info to help you along the way.

tstastny, 2023-07-10:
> The only question you did not mention was, why it did not land and aborted the landing?

Ah! Sorry, somehow missed it.

So the aircraft did a "go around" because it did not detect a distance sensor, which was set required by `FW_LND_USETER`. I'd recommend also looking at the configurable abort behavior you want additionally with this param `FW_LND_ABORT`.

When aborting landing, the behavior is to loiter above the land point. So the behavior you saw is expected.

General landing logic is well documented here:
https://docs.px4.io/main/en/flight_modes/mission.html#landing-sequence

Abort is specifically documented here:
https://docs.px4.io/main/en/flight_modes/mission.html#abort

tubeme, 2023-07-13:
Thanks a lot.

closed 2023-07-13 by tubeme