PX4/PX4-Autopilot #21788: [Bug]

toddkuen, 2023-07-02:
### Describe the bug

Error occurs for all of these:
$ make px4_sitl gazebo-classic 
$ make px4_sitl gazebo 
$ make px4_sitl jmavsim
$ make px4_sitl_default

Build on Mac OSX 13.4.1, Xcode 14.3.1
Followed: https://docs.px4.io/main/en/dev_setup/dev_env_mac.html
brew informs tbb@2020 is not supported, I am using tbb@2021
QGroundControl installed and running
cloned and installed PX4-Autopilot without errors (at least that I can see):

$ make px4_sitl gazebo-classic                                                 
[  0%] Built target events_header
[  1%] Built target flighttasks_generated
[  1%] Building C object boards/px4/sitl/src/CMakeFiles/drivers_board.dir/sitl_led.c.o
[  1%] Built target uorb_headers
[  1%] Built target output_functions_header
[  1%] Built target git_gps_devices
[  1%] Generating git version header
[  2%] Built target mode_util
[  3%] Built target parameters_xml
/Users/toddkuen/Thread/PX4/PX4-Autopilot/boards/px4/sitl/src/sitl_led.c:53:23: fatal error: a function declaration without a prototype is deprecated in all versions of C [-Wstrict-prototypes]
__EXPORT void led_init()
                      ^
                       void
1 error generated.

Fixing that I get the following error:

$ make px4_sitl gazebo-classic
[  1%] Built target events_header
[  1%] Built target flighttasks_generated
[  1%] Building C object boards/px4/sitl/src/CMakeFiles/drivers_board.dir/sitl_led.c.o
[  1%] Built target uorb_headers
[  1%] Generating git version header
[  1%] Built target git_gps_devices
[  1%] Built target output_functions_header
[  2%] Built target mode_util
[  3%] Built target parameters_xml
make[4]: *** No rule to make target `src/modules/uxrce_dds_client/dds_topics.h', needed by `events/px4.json'.  Stop.
make[3]: *** [src/lib/events/CMakeFiles/events_json.dir/all] Error 2
make[3]: *** Waiting for unfinished jobs....
[  3%] Built target actuators_json
[  3%] Linking CXX static library libdrivers_board.a
[  3%] Built target drivers_board
[  3%] Built target ver_gen
make[2]: *** [src/modules/simulation/simulator_mavlink/CMakeFiles/gazebo-classic.dir/rule] Error 2
make[1]: *** [gazebo-classic] Error 2
make: *** [px4_sitl] Error 2

There isn't much if anything on these errors in google, etc. so it's likely some kind of local to my machine config error but I have no idea what.  

I have this working on Ubuntu - the install was broken there but I was able to figure out various missing python and other components.

### To Reproduce

I would imagine this will not easily reproduce...

### Expected behavior

It should build and fire up the sim on gazebo.

### Flight Log

 N/A

### Software Version

Trying to do initial install of STIL Gazebo sim on Mac.

### Flight controller

None

### Vehicle type

None

chuangniu1, 2023-07-21:
@toddkuen Have you made any progress on this problem？

MacBook-Pro PX4-Autopilot % make px4_sitl_default jmavsim
-- PX4 version: v1.11.3-1-g21c82a4814
-- PX4 config file: /Users/Desktop/niucode/px4_open/PX4-Autopilot/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is AppleClang 14.0.3.14030022
-- The C compiler identification is AppleClang 14.0.3.14030022
-- The ASM compiler identification is Clang with GNU-like command-line
-- Found assembler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.9.17", minimum required is "3") 
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-404-g395c8d3
-- Configuring done (8.2s)
-- Generating done (1.1s)
-- Build files have been written to: /Users/chuangniu/Desktop/niucode/px4_open/PX4-Autopilot/build/px4_sitl_default
[0/739] git submodule src/drivers/gps/devices
[5/739] git submodule src/lib/ecl
[6/739] Building C object boards/px4/sitl/src/CMakeFiles/drivers_board.dir/sitl_led.c.o
**FAILED: boards/px4/sitl/src/CMakeFiles/drivers_board.dir/sitl_led.c.o** 
/usr/local/bin/ccache /Library/Developer/CommandLineTools/usr/bin/cc -DCONFIG_ARCH_BOARD_PX4_SITL -DENABLE_LOCKSTEP_SCHEDULER -D__CUSTOM_FILE_IO__ -D__PX4_DARWIN -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I/Users/chuangniu/Desktop/niucode/px4_open/PX4-Autopilot/boards/px4/sitl/src -I/Users/chuangniu/Desktop/niucode/px4_open/PX4-Autopilot/platforms/posix/src/px4/common/include -I/Users/chuangniu/Desktop/niucode/px4_open/PX4-Autopilot/build/px4_sitl_default -I/Users/chuangniu/Desktop/niucode/px4_open/PX4-Autopilot/build/px4_sitl_default/src/lib -I/Users/chuangniu/Desktop/niucode/px4_open/PX4-Autopilot/platforms/posix/src/px4/generic/generic/include -I/Users/chuangniu/Desktop/niucode/px4_open/PX4-Autopilot/platforms/common/include -I/Users/chuangniu/Desktop/niucode/px4_open/PX4-Autopilot/src -I/Users/chuangniu/Desktop/niucode/px4_open/PX4-Autopilot/src/include -I/Users/chuangniu/Desktop/niucode/px4_open/PX4-Autopilot/src/lib -I/Users/chuangniu/Desktop/niucode/px4_open/PX4-Autopilot/src/lib/matrix -I/Users/chuangniu/Desktop/niucode/px4_open/PX4-Autopilot/src/modules -I/Users/chuangniu/Desktop/niucode/px4_open/PX4-Autopilot/platforms/posix/include -I/Users/chuangniu/Desktop/niucode/px4_open/PX4-Autopilot/build/px4_sitl_default/external/Install/include -O2 -g -DNDEBUG -std=gnu11 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.3.sdk -fPIC -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -Wno-logical-op -fcolor-diagnostics -fdiagnostics-absolute-paths -Qunused-arguments -Wno-unknown-warning-option -Wno-unused-const-variable -Wno-varargs -fno-common -Wbad-function-cast -Wnested-externs -Wstrict-prototypes -MD -MT boards/px4/sitl/src/CMakeFiles/drivers_board.dir/sitl_led.c.o -MF boards/px4/sitl/src/CMakeFiles/drivers_board.dir/sitl_led.c.o.d -o boards/px4/sitl/src/CMakeFiles/drivers_board.dir/sitl_led.c.o -c /Users/chuangniu/Desktop/niucode/px4_open/PX4-Autopilot/boards/px4/sitl/src/sitl_led.c
/Users/chuangniu/Desktop/niucode/px4_open/PX4-Autopilot/boards/px4/sitl/src/sitl_led.c:53:23: fatal error: a function declaration without a prototype is deprecated in all versions of C [-Wstrict-prototypes]
**__EXPORT void led_init()
                      ^
                       void
1 error generated.**
[13/739] Generating uORB topic headers
ninja: build stopped: subcommand failed.
make: *** [px4_sitl_default] Error 1

bousborne, 2023-07-21:
@chuangniu1 I had this problem a few weeks ago. I think it has something to do with the newer SDKs on MacOS. They seem much stricter in their warning-to-error translations.

This is a "workaround", and not really a fix:
To fix it, I first installed MacOS sdk 12, but I found that somewhere, something, was actually still linking MacOS SDK 13 to my default. Therefore I still had the problem as clang would see and use 13, even though I set 12 to default. So, I ended up deleting all references to SDK 13, and it works now. But, make sure you don't have any other projects that need to be built on 13 before doing this.

You can do this either with xcode-select on the cli, or through Xcode.

Through cli do:

```
xcode-select --install  # <-- This installs sdk 12 adn 13
xcode-select -p          # <-- This will print the install location for both 12 and 13
```
cd to that location and delete all directories for sdk 13.

Then try to build again.

andrejpodzimek, 2024-02-05:
The `src/modules/uxrce_dds_client/dds_topics.h` problem definitely rings the bell, but I’m seeing it on ArchLinux, i.e. nothing OS X related. [It is mentioned](https://discuss.px4.io/t/a-question-about-px4-documents/35295/4) in a few other threads, but there doesn’t seem to be a clear explanation.

My workaround is based on looking into `src/modules/uxrce_dds_client/CMakeLists.txt`, which is where the auto-generated file is specified like this:

```cmake
add_custom_command(OUTPUT ${CMAKE_CURRENT_BINARY_DIR}/dds_topics.h
	COMMAND ${PYTHON_EXECUTABLE} ${CMAKE_CURRENT_SOURCE_DIR}/generate_dds_topics.py
		--topic-msg-dir ${PX4_SOURCE_DIR}/msg
		--client-outdir ${CMAKE_CURRENT_BINARY_DIR}
		--dds-topics-file ${CMAKE_CURRENT_SOURCE_DIR}/dds_topics.yaml
		--template_file ${CMAKE_CURRENT_SOURCE_DIR}/dds_topics.h.em
	DEPENDS
		${CMAKE_CURRENT_SOURCE_DIR}/generate_dds_topics.py
		${CMAKE_CURRENT_SOURCE_DIR}/dds_topics.yaml
		${CMAKE_CURRENT_SOURCE_DIR}/dds_topics.h.em
	COMMENT "Generating XRCE-DDS topic bridge"
)
```

Based on that↑ recipe, I simply reproduced what it tells (or should tell) `cmake` + `make` to do, but manually:

```bash
mkdir build
cd build
cmake ..
make  # <<< Fails!

python \
    ../src/modules/uxrce_dds_client/generate_dds_topics.py \
    --topic-msg-dir ../msg \
    --client-outdir ./src/modules/uxrce_dds_client/ \
    --dds-topics-file ../src/modules/uxrce_dds_client/dds_topics.yaml \
    --template_file ../src/modules/uxrce_dds_client/dds_topics.h.em

make  # <<< Works!
```

It is unclear to me why `cmake` doesn’t get this right. Everything else seems to be working fine in the (default) build after the hack / workaround.

This is on ArchLinux, up-to-date, today’s `PX4-Autopilot`, with [`ros2-arch-deps`](https://aur.archlinux.org/packages/ros2-arch-deps) installed and with [`requirements.txt`](https://github.com/PX4/PX4-Autopilot/blob/main/Tools/setup/requirements.txt) _applied_ (meaning: `sed 's/[^a-z].*$//' ../Tools/setup/requirements.txt | while read p; do paru -Sy --noconfirm --needed "python-${p}"; done` and, on top of that, `pip3 install --break-system-packages --user pyros-genmsg`, because `pyros-genmsg` is not in AUR). `cmake` version is `3.28.2`.

andrejpodzimek, 2024-02-11:
After today’s `git pull` I can’t reproduce the problem on ArchLinux any more. Looks like it’s fixed. (I can’t spot the commit that fixed it though, which is puzzling (i.e. I have no understanding of the root cause).)

mengchaoheng, 2024-03-01:
> 通过 cli 执行以下操作：
> 
> ```
> xcode-select --install  # <-- This installs sdk 12 adn 13
> xcode-select -p          # <-- This will print the install location for both 12 and 13
> ```
> 
> cd 到该位置并删除 sdk 13 的所有目录。

when I run this:
```Console
➜  PX4-Autopilot git:(jint) ✗ xcode-select --install
xcode-select: note: Command line tools are already installed. Use "Software Update" in System Settings or the softwareupdate command line interface to install updates
➜  PX4-Autopilot git:(jint) ✗ xcode-select -p
/Library/Developer/CommandLineTools
```
and 
```Console 
➜  PX4-Autopilot git:(jint) ✗ cd /Library/Developer/CommandLineTools

➜  CommandLineTools ls
Library SDKs    usr
➜  CommandLineTools cd SDKs 
➜  SDKs ls
MacOSX.sdk     MacOSX13.3.sdk MacOSX13.sdk   MacOSX14.2.sdk MacOSX14.sdk
➜  SDKs 
```
which one I should delete ? @bousborne

chuangniu1, 2024-04-22:
@mengchaoheng 
old command line version
clang --version
Apple clang version 14.0.3 (clang-1403.0.22.14.1)
Target: x86_64-apple-darwin22.5.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin

I upload clang-14.0.0 and install
link: https://developer.apple.com/download/all/

after install, and clang --version
Apple clang version 14.0.3 (clang-1403.0.22.14.1)
Target: x86_64-apple-darwin22.5.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin

then, delete SDKs ->MacOSX13

then, can make

david-rse, 2024-06-05:
This line has the incorrect relative path:
`    --client-outdir ./src/modules/uxrce_dds_client/ \`
Note that there should be another `.`:
`    --client-outdir ../src/modules/uxrce_dds_client/ \`

david-rse, 2024-06-06:
Figured out a solution for this:

Long story short add `modules__uxrce_dds_client` as a `DEPENDS` line to `add_custom_command(OUTPUT ${generated_events_px4_file}` of `src/lib/events/CMakeLists.txt`.

To diagnosis, I ran the following:
`cmake -DCMAKE_VERBOSE_MAKEFILE=ON cmake .`

And inspected the last command which failed:

`make  -f src/lib/events/CMakeFiles/events_json.dir/build.make src/lib/events/CMakeFiles/events_json.dir/build`

and noted that the `events` module was failing compilation due to `dds_topics.h` not existing. And realized that this header should be generated via python-empy templating from `dds_topics.h.em`. Meaning that the uxrce_dds_client module is needed as a prerequisite and build should precede the events module.

[image]

Long story short, add the dds project as a dependency to px4.json file that is being built. Edit `src/lib/events/CMakeLists.txt` and add the following `modules__uxrce_dds_client` as a dependency to the `generated_events_px4_file` custom command.

[image]

Run `cmake .` again and `make`
Cheers

mirusu400, 2025-04-11:
I think @david-rse 's comment is the only clear solution, but why still not fixed yet and still opened?

dakejahl, 2025-04-11:
@mirusu400 none of the core developers use MacOS so it doesn't get tested regularly. Can you open a PR to fix the issue?

mirusu400, 2025-04-12:
@dakejahl Actually I'm using Ubuntu 22.04 with docker or wsl, but I met this issue when using `cmake`, same as @andrejpodzimek did.

```bash
# I used ubuntu 22.04 docker image
# make environment normally, using https://docs.px4.io/main/en/dev_setup/dev_env.html)
mkdir build
cd build
cmake ..
make

# ...

[ 13%] Completed 'libmicroxrceddsclient_project'
[ 13%] Built target libmicroxrceddsclient_project
[ 14%] Generating parameters.xml
[ 14%] Built target parameters_xml
[ 14%] Generating git version header
[ 14%] Built target ver_gen
make[2]: *** No rule to make target 'src/modules/uxrce_dds_client/dds_topics.h', needed by 'events/px4.json'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:19164: src/lib/events/CMakeFiles/events_json.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```

btw I create PR here: #24712