PX4/PX4-Autopilot #21805: [Bug] Vehicle Local Position is not reinitialized.

fbenti, 2023-07-05:
### Describe the bug

I'm facing an issue with the `vehicle_local_position`. I run multiple offboard missions, simply taking off and landing. The altitude height is set to 1.5 m however the drone sometimes stabilizes around 4/5 meters. You can see attached the plots regarding 4 consecutive runs,

### To Reproduce

Consecutive takeoff and landing

### Expected behavior

I would expect the `vehicle_local_ position` to be initialized at every run, also because at the end of every run the XRCE bridge between the onboard computer and px4 is closed. Am I wrong? Is that expected? 

### Screenshot / Media

[image]
[image]
[image]
[image]

### Flight Log

https://logs.px4.io/plot_app?log=06ff42d2-05c0-4205-b67f-37ccff8b34da
https://logs.px4.io/plot_app?log=b709e4c2-7f4a-4ad6-b58b-18f537a92291
https://logs.px4.io/plot_app?log=91df7265-c901-44d1-be01-68cdf2c190a9
https://logs.px4.io/plot_app?log=3fa0cc55-99eb-4df6-ae20-7c7687a75b65

### Software Version

v1.14 beta

### Flight controller

PX4_FMU_v5

### Vehicle type

Multicopter

fbenti, 2023-07-05:
Is there a command to reset the vehicle_local_position estimator without the need of switching off the flight controller?

bresch, 2023-07-06:
The origin of the local frame is set at boot and shouldn't be reset. Maybe one thing you didn't consider is that local_position.z is not by definition the altitude above ground nor above home, it's the altitude above the origin (they roughly match if you boot and takeoff at the same place but can also be quite different).

If you want to takeoff at some altitude above ground, just take the current Z position and add (or subtract since it's NED) the desired altitude above ground at each takeoff in your offboard code. If you have a range finder you could also control the "dist bottom" height.

julianoes, 2023-10-19:
This has been answered and there was no follow up. Closing.