PX4/PX4-Autopilot #21810: [Bug] Navigator no longer sends fake ADSB / wrong instructions

hamishwillee, 2023-07-06:
### Describe the bug

The [navigator fake_traffic](http://docs.px4.io/main/en/modules/modules_controller.html#navigator) command is broken.

> `fake_traffic`  publishes 4 fake transponder_report_s uORB messages

If you run the command nothing happens.

The faketrigger code is still called, but all the triggers are commented out: https://github.com/PX4/PX4-Autopilot/blob/main/src/lib/adsb/AdsbConflict.cpp#L391.

This was changed in #21283 . 

### To Reproduce

Start a console (or use MAVLink console. Enter Simulator say) and enter `navigator fake_traffic`. If you uncomment the code in https://github.com/PX4/PX4-Autopilot/blob/main/src/lib/adsb/AdsbConflict.cpp#L391, rebuild, and try again, it should work. 

### Expected behavior

The command should match the behaviour - it doesn't fire off the warnings.

What I think should happen is that the test data should dynamically create some phantom adsb traffic around the vehicle just outside the default warning ranges for ADSB, using different options to the `fake_traffic()` method. This would allow testing of ADSB being displayed from vehicle on GCS.

The values should be set "just outside" the default range for warnings so that there are no default actions. The user can then modify the range slightly (we provide guidance) to make the failsafe trigger, testing other behaviour.

I think I would remove the existing "faking" code and provide instructions on how to send that same information via MAVLink for testing - i.e. add and ADSB test using MAVSDK.

### Flight Log

NA

### Software Version

PX4 main, QGC daily.

### Flight controller

Simulator, but doesn't matter

### Vehicle type

Multicopter

akkawimo, 2023-09-29:
I wasn't able to find a way around the comments. See commit https://github.com/PX4/PX4-Autopilot/pull/21283/commits/f81944efc46a16cd9d8bc5b02ed72668b5da2bf4

akkawimo, 2024-04-11:
Should be added back in https://github.com/PX4/PX4-Autopilot/pull/21663