PX4/PX4-Autopilot #21811: [Crash] Hexacopter crash caused by full power on a motor with v1.13.2

Katawann, 2023-07-06:
### Describe the bug

Our hexcopter was flying offboard when suddenly (3:15) one of the engines revved up causing the drone to go sideways at full speed and crash further away. The offboard gives velocity inputs, but I don't see the velocity points having a value that would justify full thrust from the motor.

Does anyone experienced such behavior ? I don't get why suddenly the drone received a big input like this for one of the motors... Any expert eyes would be really welcome here ! 

### To Reproduce

Not sure yet...

### Expected behavior

Drone should not receive full power on a motor like this

### Flight Log

https://logs.px4.io/plot_app?log=7743e2cd-4654-4f5f-87db-36c787df42b6

### Software Version

v.1.13.2

### Flight controller

Pixhawk 4

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

- Motors X8 from Hobbywing command with ESC 
- Raspberry Pi 3 as companion computer connected through TELEM2
- CUAV F9P RTK
- Holybro M8N
- Mauch 12-14s power supply
- 12S battery

Igor-Misic, 2023-07-06:
One motor stops spinning and FC tried to compensate for it but nothing happened.

Katawann, 2023-07-06:
Hi @Igor-Misic thank you for taking the time to analyse my issue. So it would mean the issue is in the other way around, so one motor was not turning and the others tried to compensate rather than one motor going full crazy ?

How do I know which motor is assigned to which actuator controls output ? In the graph in Flight review the actuator output 0 go full up and output 1 go full down but in your graph it is respectively the 03 and 02.

The airframe I use is the Hexacopter X
[image]

Igor-Misic, 2023-07-06:
I think we solved this in Discord but for anyone in the future here couple more things
Each motor number responds to actuator_ouptut + 1 

So Motor_1 ==  actuator_ouptut_0
So Motor_2 ==  actuator_ouptut_1
So Motor_3 ==  actuator_ouptut_2
 
[image]

What is also visible from comparing the output of the actuator with vechicle_rates_setpoint/roll is that roll went in the opposite direction.

So motor 1 failed and vechicle_rates_setpoint/roll is trying to compensate by increasing actuator_0 (motor1) and decreasing actuator_1 (motor2).