PX4/PX4-Autopilot #21820: [Crash] Loss of altitude caused by jump in Fused Altitude Estimate

Katawann, 2023-07-07:
### Describe the bug

Hello,

we just had a crash today with one of our hexacopter drone. The drone suddenly shift its position of 6m down causing the crash in vineyard. From what I see there is an unexplained change in GPS altitude, the EKF algorithm seems to trig the vertical position issue but few seconds later it still apply the altitude of GPS in the Fused Altitude Estimation

### To Reproduce

Not sure yet...

### Expected behavior

PX4 should not detect an error with the GPS altitude and rather use barometer or secondary GPS to define the altitude rather to do a big jump like this ?

### Flight Log

https://logs.px4.io/plot_app?log=852be942-403a-4299-99fc-513ae8df0f55

### Software Version

v1.13.2

### Flight controller

Pixhawk 4

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

- Motors X8 from Hobbywing command with ESC
- Raspberry Pi 3 as companion computer connected through TELEM2
- CUAV F9P RTK
- Holybro M8N
- Mauch 12-14s power supply
- 12S battery

bresch, 2023-07-11:
Sorry to hear about the crash. The reason why it jumps is because GNSS is considered as the absolute reference. We don't know is the altitude was correct before or if it was wrong and corrected itself after the jump.
I think the main issue here is rather that the drone went down to reach the setpoint (the mission WP are also absolute positions). If the altitude of the 2nd GNSS receiver didn't jump, it's true that we could leverage this. 

Otherwise, we could also try to adjust the mission items but again, we don't really know if the height was correct before or now; but we could define that if a height reset occurs during a mission we want to adjust the mission height to continue the flight with the same altitude (even if it's not the correct absolute one).

Katawann, 2023-07-11:
Thank you for checking my case @bresch ! Maybe few questions then:
- How can we define that a height reset occurs ? With the _Vertical Position Check Bit_ ?
- The drone was in _Offboard_ mode, so I could add this correction in my code when a height reset occurs but I guess it could be an issue once PX4 handles this case (as it would have double correction then) ?
- I have [another open issue](https://github.com/PX4/PX4-Autopilot/issues/21828) on the same subject but IMO PX4 does not manage properly to switch to the secondary GPS if the primary GPS gives a bad information
- In this current situation, do you think that the _Height Fusion_ feature of v1.14 could have compensate this GPS height error ?
- I feel the secondary GPS also had a change in the altitude is there a known root cause for the altitude to change suddenly on multiple satellites ? 

[image]