PX4/PX4-Autopilot #21824: [Bug] Installing PX4-Autopilot on Ubuntu22.04 Raspberry Pi 4 fails

shivam5594, 2023-07-10:
### Describe the bug

I have been following this [ROS2 user guide](https://docs.px4.io/main/en/ros/ros2_comm.html#install-px4) to install PX4 on my Raspberry Pi 4 and `ubuntu.sh` script fails with the following error
```
$ bash ./PX4-Autopilot/Tools/setup/ubuntu.sh
Ubuntu 22.04

Installing PX4 general dependencies
Hit:1 http://packages.osrfoundation.org/gazebo/ubuntu-stable jammy InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports jammy InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease
Reading package lists...
W: http://packages.osrfoundation.org/gazebo/ubuntu-stable/dists/jammy/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Reading package lists...
Building dependency tree...
Reading state information...
build-essential is already the newest version (12.9ubuntu3).
file is already the newest version (1:5.41-3).
g++ is already the newest version (4:11.2.0-1ubuntu1).
gcc is already the newest version (4:11.2.0-1ubuntu1).
make is already the newest version (4.3-4.1build1).
zip is already the newest version (3.0-12build2).
astyle is already the newest version (3.1-2build1).
cppcheck is already the newest version (2.7-1).
lcov is already the newest version (1.15-1).
libfuse2 is already the newest version (2.9.9-5ubuntu3).
ninja-build is already the newest version (1.10.1-1).
shellcheck is already the newest version (0.8.0-2).
git is already the newest version (1:2.34.1-1ubuntu1.9).
libxml2-dev is already the newest version (2.9.13+dfsg-1ubuntu0.3).
libxml2-utils is already the newest version (2.9.13+dfsg-1ubuntu0.3).
python3-setuptools is already the newest version (59.6.0-1.2ubuntu0.22.04.1).
rsync is already the newest version (3.2.7-0ubuntu0.22.04.2).
unzip is already the newest version (6.0-26ubuntu3.1).
python3-pip is already the newest version (22.0.2+dfsg-1ubuntu0.3).
python3-wheel is already the newest version (0.37.1-2ubuntu0.22.04.1).
cmake is already the newest version (3.22.1-1ubuntu1.22.04.1).
gdb is already the newest version (12.1-0ubuntu1~22.04).
python3 is already the newest version (3.10.6-1~22.04).
python3-dev is already the newest version (3.10.6-1~22.04).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Installing PX4 Python3 dependencies
Requirement already satisfied: argcomplete in ./.local/lib/python3.10/site-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 1)) (3.1.1)
Collecting argparse>=1.2
  Using cached argparse-1.4.0-py2.py3-none-any.whl (23 kB)
Requirement already satisfied: cerberus in ./.local/lib/python3.10/site-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 3)) (1.3.4)
Requirement already satisfied: coverage in ./.local/lib/python3.10/site-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 4)) (7.2.7)
Requirement already satisfied: empy>=3.3 in ./.local/lib/python3.10/site-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 5)) (3.3.4)
Requirement already satisfied: future in /usr/lib/python3/dist-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 6)) (0.18.2)
Requirement already satisfied: jinja2>=2.8 in ./.local/lib/python3.10/site-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 7)) (3.1.2)
Requirement already satisfied: jsonschema in ./.local/lib/python3.10/site-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 8)) (4.18.0)
Requirement already satisfied: kconfiglib in ./.local/lib/python3.10/site-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 9)) (14.1.0)
Requirement already satisfied: lxml in ./.local/lib/python3.10/site-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 10)) (4.9.3)
Requirement already satisfied: matplotlib>=3.0.* in ./.local/lib/python3.10/site-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 11)) (3.7.2)
Requirement already satisfied: numpy>=1.13 in /usr/lib/python3/dist-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 12)) (1.21.5)
Requirement already satisfied: nunavut>=1.1.0 in ./.local/lib/python3.10/site-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 13)) (2.1.1)
Requirement already satisfied: packaging in /usr/lib/python3/dist-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 14)) (21.3)
Requirement already satisfied: pandas>=0.21 in ./.local/lib/python3.10/site-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 15)) (2.0.3)
Requirement already satisfied: pkgconfig in ./.local/lib/python3.10/site-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 16)) (1.5.5)
Requirement already satisfied: psutil in ./.local/lib/python3.10/site-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 17)) (5.9.5)
Requirement already satisfied: pygments in /usr/lib/python3/dist-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 18)) (2.11.2)
Requirement already satisfied: wheel>=0.31.1 in /usr/lib/python3/dist-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 19)) (0.37.1)
Requirement already satisfied: pymavlink in ./.local/lib/python3.10/site-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 20)) (2.4.39)
Requirement already satisfied: pyros-genmsg in ./.local/lib/python3.10/site-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 21)) (0.5.8)
Requirement already satisfied: pyserial in ./.local/lib/python3.10/site-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 22)) (3.5)
Requirement already satisfied: pyulog>=0.5.0 in ./.local/lib/python3.10/site-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 23)) (1.0.1)
Requirement already satisfied: pyyaml in /usr/lib/python3/dist-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 24)) (5.4.1)
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 25)) (2.25.1)
Requirement already satisfied: setuptools>=39.2.0 in /usr/lib/python3/dist-packages (from -r /home/shivam/PX4-Autopilot/Tools/setup/requirements.txt (line 26)) (59.6.0)
[... 37 lines cut ...]

Just follow the [user guide](https://docs.px4.io/main/en/ros/ros2_comm.html#install-px4) 

### Expected behavior

It should complete fine

### Flight Log

N/A

### Software Version

Installation fails.

Ubuntu 22.04 LTS
Raspberry Pi 4 8GB

### Flight controller

NA

### Vehicle type

None

shivam5594, 2023-07-10:
`make px4_sitl` works fine with this error.

Found this forum from 2015 which says that g++-multilib is not available anymore for raspberry pi. [Read this](https://forums.raspberrypi.com/viewtopic.php?t=101187)

If this is a known issue and someone can confirm, I'd like to request to update that script.

junwoo091400, 2023-07-10:
> Found this forum from 2015 which says that g++-multilib is not available anymore for raspberry pi. [Read this](https://forums.raspberrypi.com/viewtopic.php?t=101187)

Yes. That is the issue, but removing this dependency will also not be the solution as this is only happening because the g++ multilib isn't compiled and distributed for raspberry pi targets.

Have you found any other similar posts, in PX4 discuss forum or issues?

shivam5594, 2023-07-11:
@junwoo091400  [This](https://github.com/PX4/PX4-Autopilot/issues/18180) is the closest one I could find