PX4/PX4-Autopilot #21825: [Bug] Unable to Download Parameter into GCS in Mainland China

vincentpoont2, 2023-07-10:
### Describe the bug

QGC would get stuck at "loading" in Mainland China. 
[image]

- The loading would resume if an VPN is used, or internet is disconnected.
- Tested with QGC June 7th Daily build & stable 4.2.
- It only happens to master FW, does not happen to v1.13.3 stable, v1.14 beta. 
- No issue with flashing PX4 FW
- Tested with Pixhawk 6C & Kakute H7 V2.
- Except for the parameters downloading, everything runs well. You can see the attitude, flight mode ,all the telemetry is updating

@dagar @mrpollo Any idea? Is there difference in the server location for 1.14 or something related? 

https://github.com/PX4/PX4-Autopilot/assets/46874772/edc89419-d343-4ee2-82e9-37a52de55452

### To Reproduce

Connect FC to QGC in Mainland China. (tested in Shenzhen)

### Expected behavior

QGC to download parameter and work properly.

### Screenshot / Media

A user reported that this issue also occur in Mission Planner with PX4 Master. 
[image]

Discord thread. https://discord.com/channels/1022170275984457759/1097102345537597490/1103876488794734733

### Flight Log

N/A

### Software Version

Master

### Flight controller

Pixhawk 6C, CUAV-x7pro, holybro-kakiteh7v2, etc

### Vehicle type

None

junwoo091400, 2023-07-10:
Could you enable the Parameter related debug entries (e.g. ParameterManagerLog, ParameterManagerDebugCacheFailureLog, etc) and gather the debug output of the QGC?

You can check the documentation here for reference: https://docs.qgroundcontrol.com/master/en/SettingsView/console_logging.html

vincentpoont2, 2023-07-14:
@junwoo091400 I am not familar with these parameters. We enabled the `ParameterManagerLog` & `ParameterManagerDebugCacheFailureLog`. Not sure if one can find some useful detail with these logs. 

[Pixhawk 6C_Master20230605.txt](https://github.com/PX4/PX4-Autopilot/files/12046187/Pixhawk.6C_Master20230605.txt)
[Pixhawk 6C_Stable 1.13.3.txt](https://github.com/PX4/PX4-Autopilot/files/12046188/Pixhawk.6C_Stable.1.13.3.txt)

danielhonies, 2023-08-09:
@vincentpoont2 maybe you could check with something like wireshark on what requests are sent?

julianoes, 2023-10-20:
Does it work in flight mode without any internet?