PX4/PX4-Autopilot #21828: [Bug]

Katawann, 2023-07-10:
### Describe the bug

One of our hexacopter had a crash today. The root cause is known and it is because cable connected to GPS antenna unscrewed causing the loss of the satelittes... The Stabilized mode is also available on the RC but the pilot did not have the reflex to switch to it with all the stress...

My question is more related to the [Position failsafe](https://docs.px4.io/main/en/config/safety.html#position-gps-loss-failsafe) not triggered. I don't understand why in this case where there is absolutely no more satellites, PX4 does not switch to Altitude or Stabilized by itself ? I also have a secondary GPS working properly so there is no failsafe to use this one if the primary is failing ?

I also had [another issue](https://github.com/PX4/PX4-Autopilot/issues/21820#issue-1793617024) few days ago where PX4 did not understand a bad information from the GPS so I'm wondering if I'm missing some parameters or if PX4 does not handle this situation correctly

### To Reproduce

Can maybe be reproduced in simulation with a loss of satellites or in real by interrupting the cable between antenna and ublox F9P

### Expected behavior

PX4 should detect a loss of satellites and trigger the [Position (GPS) Loss Failsafe](https://docs.px4.io/main/en/config/safety.html#position-gps-loss-failsafe) mode and switch to _Altitude_ or _Stabilized_

PX4 should use the secondary GPS if the primary is failing 

### Flight Log

https://logs.px4.io/plot_app?log=a09a04f9-beef-403b-9a33-5cdf0df3097b

### Software Version

v1.13.2

### Flight controller

Pixhawk 4

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

- Motors X8 from Hobbywing command with ESC
- Raspberry Pi 3 as companion computer connected through TELEM2
- CUAV F9P RTK
- Holybro M8N
- Mauch 12-14s power supply
- 12S battery

Katawann, 2023-07-13:
Here is another flight with another drone where we had loss of satellites but no Position failsafe triggered...
https://logs.px4.io/plot_app?log=b70172e9-884d-4d87-b4ae-a04f0f4e0a04

bresch, 2023-07-20:
The failsafe wasn't triggered because the EKF2 never stopped using the GNSS data, even if it reported a loss of fix. This has been patched in 1.14 already: https://github.com/PX4/PX4-Autopilot/pull/20686

Regarding the switch to another GNSS receiver, there is currently no logic implemented for this. An automatic failover only occurs if the data stops completely.

Katawann, 2023-07-20:
Thank you @bresch for checking my case ! I close this issue and leave the discussion open for later regarding the logic to handle to switch to the secondary GPS if the primary fails

closed 2023-07-20 by Katawann