PX4/PX4-Autopilot #21836: [Bug] High RAM usage on NXP FMUK66 on main

dirksavage88, 2023-07-11:
### Describe the bug

High ram usage (maxed) on main using the FMUK66 by Nxp.

### To Reproduce

This is with uavcan enabled and active connections to a uavcan gps+mag. Could not open a system console or console commands not recognized due to max ram usage. Nothing else out of the ordinary running (see screenshot)

### Expected behavior

Could not replicate on stable/1.13 where ram usage was not causing issues.

### Screenshot / Media

[image]
[image]

### Flight Log

Will send log on discord at request (dm)

### Software Version

Main

### Flight controller

FMUK66

### Vehicle type

Multicopter

dirksavage88, 2023-07-14:
So according to @PetervdPerk-NXP this is a known issue with uavcan enabled due to listener expansion with no known solution as of yet. Perhaps make uavcan selective of which listeners/topics?

dirksavage88, 2023-07-26:
Thanks @junwoo091400 for updating the title now that we know the exact cause of the high ram usage on this board

henrykotze, 2023-08-04:
I dont have that NXP board but try this PR whether it helps to reduce the RAM usage.
https://github.com/PX4/PX4-Autopilot/pull/21928

dirksavage88, 2023-08-08:
I'll give it a shot @henrykotze.

In the mean time I disabled mavlink on the 2nd telem port and that helped reduce RAM usage.

mrpollo, 2023-08-08:
Hey @PetervdPerk-NXP have you experience this before? Have any helpful tips for us?

henrykotze, 2023-08-08:
@dirksavage88 you will just need to configure the board via the 'make <boardname> boardconfig' command and disable all uavcan related subscribers and controllers.

dirksavage88, 2023-09-19:
[image]

About 90%, and I disabled all the uavcan subscribers except for GPS (which I use over DroneCAN), optical flow (I have a can flow sensor I use from time to time) and rangefinder. Serial over telem is back to default.

dirksavage88, 2023-09-29:
Fixed in #21928

closed 2023-09-29 by dirksavage88