PX4/PX4-Autopilot #21837: [Bug] REQUIRED process [sitl-1] has died!

AlexWUrobot, 2023-07-11:
### Describe the bug

Install the lastest MAVROS and PX4-autopilot in Ubuntu 18.04 (python 2.7)
I placed the PX4-autopliot under root, instead of any worksapce.
When running `roslaunch px4 mavros_posix_sitl.launch `
get error like this "=REQUIRED process [sitl-1] has died!"

[image]

How to reproduce
========

ridley2@ridley2-Alienware-Aurora-R12:~/PX4-Autopilot$ roslaunch px4 mavros_posix_sitl.launch
... logging to /home/ridley2/.ros/log/83d21e16-2010-11ee-b02d-cc6b1e18f8d5/roslaunch-ridley2-Alienware-Aurora-R12-20823.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ridley2-Alienware-Aurora-R12:38571/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /mavros/camera/frame_id: base_link
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/horizontal_fov_ratio: 1.0
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/distance_sensor/sonar_1_sub/vertical_fov_ratio: 1.0
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
[... 220 lines cut ...]
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/simulation/gazebo-classic/sitl_gazebo-classic
roslaunch px4 posix_sitl.launch
```
[ref](https://docs.px4.io/main/en/simulation/ros_interface.html)

### Expected behavior

It should roslaunch successfully without error.
It should publish mavros and iris (a drone model) in the gazebo world.

### Flight Log

no log

### Software Version

Ubuntu 18.04

### Flight controller

None

### Vehicle type

None

AllenHuangDongyue, 2023-07-31:
same #21911 @lucasdemarchi @NaterGator @JamesHarrison @lfelipe

JohnTGZ, 2023-11-17:
I realised the error for PX4 SITL originates from line 324 to 334 of the file "platforms/posix/src/px4/common/main.cpp", from commit 2110eae25d32ef102c5e44f2590fd7f9d65fc44d:
```cpp
		if (test_data_path != "") {
			const std::string required_test_data_path = "./test_data";

			if (!dir_exists(required_test_data_path)) {
				ret = symlink(test_data_path.c_str(), required_test_data_path.c_str());

				if (ret != PX4_OK) {
					return ret;
				}
			}
		}

```
For some reason, the test_data_path is not empty when spawning a PX4 SITL instance. As a result, the return code will not be PX4_OK and it will return -1, ending the program.

nikhil-sethi, 2024-03-06:
For me this happened when I moved the PX4-Autopilot directory somewhere else and recompiled it in the new place. The folder `~/.ros/test_data` is a symbolic link to `PX4-Autopilot/test_data` which doesn't get updated after recompiling. I just updated the link and things worked again.

newawesom, 2024-05-16:
@nikhil-sethi I have the same problem when I moved the directory,so how to updated the link?

newawesom, 2024-05-16:
> @newawesom You can use the `ln` command (in linux) to update the symbolic link from the source directory (`~/.ros/test_data`) to the target directory (wherever your new test data folder is).
> 
> ```
> ln -s <target_directory> <source_directory>
> ```
> 
> you may also have to use `-f` flag to force the update if the link already exists.

@nikhil-sethi Thanks for reply!Acutually,I deleted test_data directory in PX4-Autopilot and it worked after recompiling.

heleidsn, 2025-04-19:
> [@nikhil-sethi](https://github.com/nikhil-sethi) Thanks for reply!Acutually,I deleted test_data directory in PX4-Autopilot and it worked after recompiling.

Worked for me~