PX4/PX4-Autopilot #21880: [Bug]

WeiyiKong-0419, 2023-07-25:
### Describe the bug

I meet a problem when make px4_sitl_default gazebo
It seems like related to the missing of something

### To Reproduce

Just make px4_sitl_default gazebo
Then I see this in Terminal
~/PX4-Autopilot$ make px4_sitl_default gazebo
[0/917] git submodule src/drivers/gps/devices
[2/917] git submodule src/modules/mavlink/mavlink
[7/917] git submodule Tools/simulation/gazebo-classic/sitl_gazebo-classic
[12/917] git submodule src/modules/uxrce_dds_client/Micro-XRCE-DDS-Client
[18/917] Performing configure step for 'sitl_gazebo-classic'
-- install-prefix: /usr/local
-- cmake build type: RelWithDebInfo
-- Using C++17 standard
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.58") found components: system thread filesystem 
-- Found DART: /usr/include (Required is at least version "6.6") found components: dart 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time 
-- Looking for ignition-math6 -- found version 6.13.0
-- Searching for dependencies of ignition-math6
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Looking for ignition-math6 -- found version 6.13.0
-- Looking for ignition-transport8 -- found version 8.3.0
-- Searching for dependencies of ignition-transport8
-- Found Protobuf: /usr/local/lib/libprotobuf.so;-lpthread (found suitable version "3.6.1", minimum required is "3") 
-- Checking for module 'uuid'
--   Found uuid, version 2.34.0
-- Looking for ignition-msgs5 -- found version 5.10.0
-- Searching for dependencies of ignition-msgs5
-- Looking for ignition-math6 -- found version 6.13.0
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 6.2.0
-- Looking for ignition-msgs5 -- found version 5.10.0
-- Looking for ignition-common3 -- found version 3.14.2
-- Searching for dependencies of ignition-common3
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- Searching for <ignition-common3> component [graphics]
-- Looking for ignition-common3-graphics -- found version 3.14.2
-- Searching for dependencies of ignition-common3-graphics
-- Looking for ignition-math6 -- found version 6.13.0
-- Looking for ignition-fuel_tools4 -- found version 4.6.0
-- Searching for dependencies of ignition-fuel_tools4
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.7.4
-- Checking for module 'yaml-0.1'
--   Found yaml-0.1, version 0.2.2
-- Checking for module 'libzip'
--   Found libzip, version 1.5.1
-- Looking for ignition-common3 -- found version 3.14.2
-- Looking for ignition-math6 -- found version 6.13.0
-- Looking for ignition-msgs5 -- found version 5.10.0
-- Building klt_feature_tracker without catkin
-- Building OpticalFlow with OpenCV
[... 73 lines cut ...]
ninja: build stopped: subcommand failed.
FAILED: external/Stamp/sitl_gazebo-classic/sitl_gazebo-classic-build 
cd /home/weiyikong/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic && /usr/bin/cmake --build /home/weiyikong/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic -- -j 8
ninja: build stopped: subcommand failed.
make: *** [Makefile:232: px4_sitl] Error 1

### Expected behavior

Emmm, It should be Ok in gazebo

### Flight Log

Nope

### Software Version

The newest one

### Flight controller

Not hardware. It's simulation

### Vehicle type

None

Jaeyoung-Lim, 2023-08-02:
@WeiyiKong-0419 Wrong protobuf version?

julianoes, 2023-10-20:
No response from OP, closing

closed 2023-10-20 by julianoes