PX4/PX4-Autopilot #21895: [Crash] Yaw estimate jumps suddenly, causing several crashes with Pixhawk 6C and PX4 (1.13.3 and 1.13.2)

DanHert, 2023-07-27:
### Describe the bug

We have recorded a large number of crashes with the new Pixhawk 6C and 6X which seem to be caused by a sudden large jump in the Yaw estimate. 

I have recorded two instances here: 
https://review.px4.io/plot_app?log=8b05047d-cc5b-4b7c-ba2e-c84877f04a19
- Mission mode, then switch to Position mode, after which the yaw estimate jumped suddenly and the UAV went unstable, resulting in a crash.

https://review.px4.io/plot_app?log=d6b953ae-1128-4e75-b663-4468ddb8d614
- Only flying in position mode, hovering in place, when suddenly yaw estimate jumps, causing a crash

Both these flights were with Pixhawk 6C + Holybro M8N, running the 1.13.3 stable release, with only one modification - a custom airframe (which is a copy of the Octacopter, just with reversed rotor direction)

I am also enclosing another case: https://review.px4.io/plot_app?log=613f484c-1fdb-4d39-9bb0-8b5e12009187
- Here we fly a quadcopter with Pixhawk 6C and 1.13.2, flying in offboard mode, Yaw estimate jumps by ~50° and the drone goes unstable (takeover to Position mode and landing)

We experienced many more crashes and sudden jumps in Yaw estimate, I can dig for more logs if you need me to.

### To Reproduce

Just flying the UAV in  any mode (Manual, Altitude, Position, Mission, Ofboard) and the jump happens spontaneously.
- Only happens with Pixhawk 6C or 6X running 1.13.2 or  1.13.3

### Expected behavior

Stable flight of the UAVs

### Screenshot / Media

case 1:
[image]

case 2:
[image]

### Flight Log

https://review.px4.io/plot_app?log=8b05047d-cc5b-4b7c-ba2e-c84877f04a19

https://review.px4.io/plot_app?log=d6b953ae-1128-4e75-b663-4468ddb8d614

https://review.px4.io/plot_app?log=613f484c-1fdb-4d39-9bb0-8b5e12009187

### Software Version

1.13.3 or 1.13.2

### Flight controller

Pixhawk 6C and Pixhawk 6X

### Vehicle type

Multicopter

DanHert, 2023-07-27:
Sorry, the third log link is incorrect, this is the one I am writing about - https://review.px4.io/plot_app?log=d96c98a6-f429-429d-9219-94173580b73f

DanHert, 2023-07-27:
Here are two more cases, even with some NaNs here and there:
https://logs.px4.io/plot_app?log=ff11fe2a-6222-4cd3-a18b-28cb6f5468e8
https://logs.px4.io/plot_app?log=fe5cdac3-db69-477f-bfcf-b966cd230b1b

bresch, 2023-07-27:
Hi, so the large jump occurs because the magnetic heading is unusable (the "Norm of mag field" in the plot below is supposed to be a flat line at around 0.4 Gauss). Something disturbing the mag field is too close to the magnetometer (that is inside the GNSS module). The GNSS module needs to be as far as possible from the EMI sources (power cables, motors, ESCs, camera, power module, ...).
[image]

The reset just makes the drone move a bit and it seems that the controllers are not tuned, this is why it starts a diverging oscillation. I would suggest to follow this guide and start by tuning the rate and attitude loops in stabilized mode before continuing: https://docs.px4.io/main/en/config_mc/pid_tuning_guide_multicopter_basic.html

What kind of drone is it? How big is it? Can you share a picture?

DanHert, 2023-07-27:
Hi, it is a large octacopter (18 inch propellers).
In this particular case, we were carrying a heavy payload (12kg) so that would explain higher interference with the magnetometer, as there is more current flowing. The GPS module is not very far, from the power cables, so interference is very much possible.

This flight: https://review.px4.io/plot_app?log=8b05047d-cc5b-4b7c-ba2e-c84877f04a19
was also with the same drone, but without the extra payload, so the Norm is ~1, which is still higher than the 0.4 you mention.

But what about the other 3 flight?
All were flown with a 13 inch-prop quadcopter (X500 size)
https://review.px4.io/plot_app?log=d96c98a6-f429-429d-9219-94173580b73f (Pixhawk 6C + 1.13.3)
https://logs.px4.io/plot_app?log=fe5cdac3-db69-477f-bfcf-b966cd230b1b (Pixhawk 6X + 1.13.2)
https://logs.px4.io/plot_app?log=ff11fe2a-6222-4cd3-a18b-28cb6f5468e8 (Pixhawk 6X + 1.13.2)

bresch, 2023-07-27:
> so the Norm is ~1, which is still higher than the 0.4 you mention.

Yes, it's still not great. Good mag data looks like this:
[image]

> But what about the other 3 flight?

The rate/attitude tuning is better on this vehicle and that's why it's not crashing. Regarding the mag field, the norm looks a bit better but the deviation still look quite large.

DanHert, 2023-07-28:
Hey! I am back with some testing results. 
It is raining outside, do I made a setup which I can test indoors. I 3D printed some paddles to replace propellers and induce some drag, so that I can simulate spinning the motors without actually flying(this setup draws 75A at full throttle, so good enough).

The drone looks like this:
[image]
There is an M8N GPS antenna with a magnetometer, and Pixhawk 6X is located right bellow the Intel NUC computer in the middle, so it sits on top of the power distribution board.

This is the first test, with default settting on FW 1.13.2:
https://review.px4.io/plot_app?log=d776a6ca-b4ae-4901-97e3-5d227c00b1fd
[image]
- I can see clear interference with the magnetometer and there is a drift in Yaw (expected)
- But when I check the magnetometers in the Mavlink Inspector, I can see that only the internal magnetometer is affected, the external one in the GPS antenna is fine

So, next test, I set these parameters:
- `CAL_MAG0_PRIO=Min` (was at medium)
- `CAL_MAG1_PRIO=Max` (was at high)
I hoped that this would mean that the estimator should prioritize the external mag. Results:
https://review.px4.io/plot_app?log=566945df-eab2-4b4f-ad34-b15e7c91bbc7
[image]
- still horrible, looks like the internal magnetometer is still used

So next I changed:
- `SENS_MAG_MODE=Publish primary magnetometer` (was at publish all magnetometers)
This has finally resolved the issue and now it looks like the external magnetometer is used:
https://review.px4.io/plot_app?log=c37182b4-9f6b-4014-a640-506f60abdbbe
[image]

 So, why does the default setup prefer the internal magnetometer, when an external one is present? We have never had to solve an issue like this with older versions of Pixhawk and the firmware.

bresch, 2023-07-31:
Nice! That's much better.
The reason why it was using the internal mag is that is't actually using...both. Your setup was apparently in "multi-ekf" mode where each estimator instance is using a different mag and IMU, meaning that some instances were using the internal mag and others the external one.
By selecting `SENS_MAG_MODE=Publish primary magnetometer`, all EKF instances are now using the primary magnetometer (external). See [here](https://docs.px4.io/main/en/advanced_config/tuning_the_ecl_ekf.html#running-a-single-ekf-instance) for more details.
As a note, we changed the defaults in v1.14 to prevent from having the internal mag used in the primary EKF instance.

junwoo091400, 2023-08-11:
> * `CAL_MAG0_PRIO=Min` (was at medium)
> 
> * `CAL_MAG1_PRIO=Max` (was at high)

I am always confused as to which MAG instance is actually internal, and which is external. However, I am aware that higher priority is given to an external magnetometer by default (detectable via which SPI/I2C bus it is on), so I guess it's a bit irrelevant for this situation (as it is EKF2 preferring internal mag).

However, I was surprised that @DanHert you had to re-arrange the sensor priority, as I *believe it should be set automatically as explained above

bresch, 2023-08-28:
Looks like the problem is solved!

closed 2023-08-28 by bresch

jorgeneo560, 2023-09-28:
i am having the same problem but with ardupilot in a plane, but i am also using pixhawk 6c with the m8n