PX4/PX4-Autopilot #21902: Support EKF2_GPS_POS_* for Multiple GPS

AlexKlimaj, 2023-07-27:
### Describe problem solved by the proposed feature

Currently the EKF2_GPS_POS_* params only apply to vehicle_gps when it gets used in the EKF.

### Describe your preferred solution

I propose we move these params to an offset in the GPS driver.

### Describe possible alternatives

Do we need these offsets at all?

dagar, 2023-07-27:
We could finish this one pretty quickly now. https://github.com/PX4/PX4-Autopilot/pull/19402

bresch, 2023-07-28:
> Do we need these offsets at all?

Yes, we definitely do. You can see that they have a huge effect on the velocity estimate when the GNSS antenna is far from the C.G. Since the measurement contains the tangential velocity at the location of the antenna in addition to the C.G. velocity, this can make the velocity loop oscillating wildly.

AlexKlimaj, 2023-11-30:
/bounty $100

pqcfox, 2024-01-07:
This seems fun to work on! Going to start an attempt today. 

/attempt #21902

jacanidjar, 2026-01-22:
/attempt #21902

tani125, 2026-02-01:
/attempt #21902

bitdeckai, 2026-02-14:
I am taking this bounty issue and starting implementation. I will share design notes and a first PR shortly.

closed 2026-02-16 by dakejahl