PX4/PX4-Autopilot #21903: [Bug] Gazebo SITL not starting (gz_bridge crash)

andrii619, 2023-07-27:
### Describe the bug

When trying to run px4 SITL with Gazebo the simulator wont start. I'm using px4 1.14 with Gazebo garden. When I run make px4_sitl gz_x500 the sitl cant communicate with Gazebo. 
I found that the problem is that gz_bridge is waiting for communication with gazebo and if not communication happens gz_bridge crashes. This happens when Gazebo is not running yet when gz_bridge is launched. A simple solution to the problem is to add a small delay to the gz_bridge launch script. When edit the file ROMFS/px4fmu_common/init.d-posix/px4-rc.simulator on line 148 and add a delay the problem is fixed:
`	elif [ -n "${PX4_SIM_MODEL}" ] && [ -z "${PX4_GZ_MODEL_NAME}" ] && [ -z "${PX4_GZ_MODEL}" ]; then

		echo "WARN  [init] PX4_GZ_MODEL_NAME or PX4_GZ_MODEL not set using PX4_SIM_MODEL."
		echo "sleeping "
		sleep 15

		if gz_bridge start -m "${PX4_SIM_MODEL#*gz_}" -w "${PX4_GZ_WORLD}" -i "${px4_instance}"; then
` 

However, this is not the right solution. Someone should fix the gz_bridge to wait on Gazebo to start. 

### To Reproduce

make px4_sitl gz_x500

### Expected behavior

The Gazebo simulator is able to communicate with px4 sitl

### Screenshot / Media

[image]

### Flight Log

N/A

### Software Version

PX4: 1.14
Gazebo: Garden
Ubuntu: 22.04

### Flight controller

SITL

### Vehicle type

None

TanguyTheOtter, 2023-08-10:
I have the same problem! Did you find a workaround?

jmackay2, 2023-10-05:
I had this issue. It appears to be the same one as #22148 . The current resolution people have found is increasing some sleep time to 5 seconds. It doesn't seem like a final solution, but the workaround currently works for me and a few others.