PX4/PX4-Autopilot #21915: Enabling camera capture for VIO applications

Jaeyoung-Lim, 2023-08-02:
### Describe problem solved by the proposed feature

Pixahakws already have all the necessary GPIOs to do camera synchronization with IMUs for VIO applications.

However, it seems that the current camera capture system is built for a specific use case of mapping with PPK, which is making the following assumptions
- Camera images are always triggered through a trigger signal. This assumption excludes the use of free-running cameras
- GPS is always available when taking images.

The following image shows a free-running camera connected to the FMU, where only the capture signal is connected. The `camera_capture` driver correctly captures the images when the camera is running.

[image]

However, the capture signals are never logged since there are no trigger signals.

Time synchronization of trigger signals is not limited to video, but also lidar/radar sensors. Having a general framework for enabling such sensors integrated to the fmu will help integrating these sensor applications with the fmu.

### Describe your preferred solution

A way of integrating a capture signal that is not necessarily triggered by a trigger signal, and still be able to later synchronize camera images with imu time stamps offboard

One of the use cases would be using a free running camera (e.g. UVC) and be able to synchronize imus with the images. This would enable visual-inertial sensor data that can be utilized for advanced applications such as VIO(Visual Inertial Odometry).

### Describe possible alternatives

- We can try to generalize the capture interface all together to be used outside for any capture signal available

### Additional context

- This problem was briefly discussed in the dev call last week

tiralonghipol, 2024-11-11:
@Jaeyoung-Lim @dagar 
hei guys, what is the status of this?

Jaeyoung-Lim, 2024-11-11:
@tiralonghipol It is as far as written in the PR.