PX4/PX4-Autopilot #21919: Support custom microXRCE profile or discovery server IP

dev10110, 2023-08-03:
### Describe problem solved by the proposed feature

When using ROS2 with multiple agents and computers, the DDS simple discovery protocol adds a lot of networking overhead. 
I would like to be able to specify an IP address of the discovery server, so that the amount of communication is minimized. 

I tried to figure out how to do this, but it seems like the participant profile is hardcoded into the PX4 source code. I would be willing to implement this feature, but I need some advice on how to setup the profile inside PX4-code. 

Refs:
- https://docs.ros.org/en/humble/Tutorials/Advanced/Discovery-Server/Discovery-Server.html

### Describe your preferred solution

Provide a PX4 parameter that decides whether the discovery server is used, and if so, specify the IP address of the discovery server.   

### Describe possible alternatives

Perhaps the XML of the participant can a separate file (similar to how mixers used to be done) that can be loaded onto the sd card? I dont like this option much, but its possible I suppose.

mrpollo, 2023-08-03:
Hey @beniaminopozzan is this something you can help us with?

dagar, 2023-08-03:
If possible I'd like to avoid the full xml embedded side, but otherwise if anything it could be off by default. The core use case we should optimize for is when you have an embedded flight controller (or PX4 SITL) talking to ROS 2 on Linux right next to it.

bkueng, 2023-08-04:
@dev10110 this is already possible. Make sure to start `uxrce_dds_client` [with the `-c` flag](https://github.com/PX4/PX4-Autopilot/blob/main/src/modules/uxrce_dds_client/uxrce_dds_client.cpp#L746), then you can customize fastdds on the agent side with a profile.
Create an xml:
```
<?xml version="1.0" encoding="UTF-8" ?>
<!-- DDS configuration for the MicroXRCEAgent -->
<dds>
    <profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles" >
        <transport_descriptors>
            <transport_descriptor>
                <transport_id>custom_udp_transport</transport_id>
                <type>UDPv4</type>
                <interfaceWhiteList><address>172.17.0.1</address></interfaceWhiteList>
            </transport_descriptor>
        </transport_descriptors>
        <!-- "px4_participant" is used by PX4 to reference this configuration -->
        <participant profile_name="px4_participant" is_default_profile="true">
            <rtps>
                <userTransports>
                    <transport_id>custom_udp_transport</transport_id>
                </userTransports>
                <useBuiltinTransports>false</useBuiltinTransports>
            </rtps>
        </participant>
    </profiles>
</dds>
```
Then do `export FASTRTPS_DEFAULT_PROFILES_FILE=xy.xml` and start the agent.

dev10110, 2023-08-04:
@bkueng and others, 

Thanks for getting back to me. I had missed the -c flag when I first looked into this, so it's cool so see that it's there. 

However the main thing I want to do is to specify the discovery server ip and port, i.e. similar to 

```

<?xml version="1.0" encoding="UTF-8" ?>
 <dds>
     <profiles xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
         <participant profile_name="super_client_profile" is_default_profile="true">
             <rtps>
                 <builtin>
                     <discovery_config>
                         <discoveryProtocol>SUPER_CLIENT</discoveryProtocol>
                         <discoveryServersList>
                             <RemoteServer prefix="44.53.00.5f.45.50.52.4f.53.49.4d.41">
                                 <metatrafficUnicastLocatorList>
                                     <locator>
                                         <udpv4>
                                             <address>127.0.0.1</address>
                                             <port>11811</port>
                                         </udpv4>
                                     </locator>
                                 </metatrafficUnicastLocatorList>
                             </RemoteServer>
                         </discoveryServersList>
                     </discovery_config>
                 </builtin>
             </rtps>
         </participant>
     </profiles>
 </dds>

```

Where I specify a discovery protocol, ip and port. 

If I launch the code with -c flag, it will only have any impact `#if defined(UXRCE_DDS_CLIENT_UDP)` which means when I'm doing a serial connection with my offboard Linux computer it won't respect the custom participant right? 

Ps, I'm not even sure if there is a need to specify the discovery protocol when using the serial communication with the offboard computer.

Yesterday I was able to implement some additional xml logic so allow for the discovery server to be set, and it works in gazebo sims, but not sure if it works in hardware yet.

bkueng, 2023-08-07:
> If I launch the code with -c flag, it will only have any impact #if defined(UXRCE_DDS_CLIENT_UDP) which means when I'm doing a serial connection with my offboard Linux computer it won't respect the custom participant right?

> Ps, I'm not even sure if there is a need to specify the discovery protocol when using the serial communication with the offboard computer.

It should work the same way. But you're right, we need to take it out if the `#if defined(UXRCE_DDS_CLIENT_UDP)`.

dev10110, 2023-08-08:
Thanks @bkueng - I'll try this out soon and let you know.

abhi-eve, 2024-06-28:
Hey @dev10110 did this work?

abhi-eve, 2024-07-03:
@bkueng Hello, 
I tried the exact same thing on hardware with serial connection with PX4, I can see the topics listed when i do `ros2 topic list` and also can see the publisher count as 1 when I do `ros2 topic info` but I am not able to echo it or subscribe to it. Everything works as expected when used with PX4 SITL and UDP. 

Anything I am missing here? Thank you

abhi-eve, 2024-07-10:
UPDATE: 
For whoever facing the same issue, I managed to get this working via serial transport. However, I had to make these two changes. 

1. Set `UXRCE_DDS_PTCFG` PX4 parameter to `Custom participant` (https://docs.px4.io/main/en/advanced_config/parameter_reference.html#:~:text=8888-,UXRCE_DDS_PTCFG,-(INT32))

2. Start the `uxrce_dds_agent` with `-r` flag and not -`-c` flag with the XML file mentioned by @dev10110  above, with `px4_participant` as profile name.