PX4/PX4-Autopilot #21920: [Bug] Multiple loss of connection with the RC lead to crash on v1.13.2

Katawann, 2023-08-03:
### Describe the bug

Hello,

we just had a crash today with one of our hexacopter drone. The drone has lost connection with the remote control on several occasions during the flight. I don't yet know the root cause of these losses, but I wanted to point out the behavior of PX4 when a loss of connection with the RC happen.

We set up the failsafe that automatically switches to Hold mode after one second after a loss of RC. But the issue is that the consecutive loss of connection caused the drone to switch from Hold to Stabilized mode several times, making difficult to control the drone (the pilot choose the Stabilized mode, thinking there was a loss of satellites)

[image]

### To Reproduce

Enable failsafe to Hold mode when RC loss happen

### Expected behavior

In this case, wouldn't it be safer to remain in the failsafe mode once the connection with the RC has been re-established?

### Flight Log

https://logs.px4.io/plot_app?log=8846a52a-eead-4e24-a30d-72c0f22f5f7c

### Software Version

v1.13.2

### Flight controller

Pixhawk 6C

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

- Motors X8 from Hobbywing command with ESC
- Raspberry Pi 3 as companion computer connected through TELEM2
- CUAV F9P RTK
- Holybro M8N
- Mauch 12-14s power supply
- 12S battery
- Remote controller H16 from Skydroid

Tris2, 2023-08-24:
Hi Katawann, this is a very interesting case. I was wondering if you were able to reproduce this bug in simulation?

julianoes, 2023-10-19:
[image]

So what was the flight mode set to? Any idea why it would show Stabilized in the manual input here?

Katawann, 2023-10-20:
Thank you for checking my issue.

So before 7:17 it was set to Position and starting from 7:17 the pilot switches the manual input to Stabilized and did not change it anymore after that. So PX4 switched from Loiter (failsafe when no RC) to Stabilized (mode chose by the pilot) repeatedly. 
[image]