PX4/PX4-Autopilot #21927: [Bug] Beta PX4 1.14 QGC missing flight mode options

benjinne, 2023-08-04:
### Describe the bug

[image]
Using the RC1 1.14 and the latest stable QGC, the flight modes setup page only offers the mode settings and it appears the other switch settings are missing like arm, kill, etc.

### To Reproduce

I'm running QGC on ubuntu 22.04 using the appimage and this error shows up when I select the flight modes tab:
`qrc:/qml/PX4SimpleFlightModes.qml:36: ReferenceError: _switchFactList is not defined`

Flight controller: ARK FMUv6x
Airframe: Generic Standard Plane

### Expected behavior

Switch settings should be available outside of modifying parameters 

### Flight Log

N/A

### Software Version

nsh: sysinit: fopen failed: No such file or directory

NuttShell (NSH) NuttX-11.0.0
nsh> ver all
HW arch: ARK_FMU_V6X
HW type: ARKV6X000000
HW version: 0x000
HW revision: 0x000
PX4 git-hash: 5359fe59e507896dace176d7fd8f9b0d00c2e3b6
PX4 version: 1.14.0 c0 (17694912)
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: 3f77354c0dc88793a47ff3b57595195ab45f7ba9
Build datetime: Jul 27 2023 18:33:08
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000600000000333039383130511800220032
MCU: STM32H7[4|5]xxx, rev. V
nsh> 

### Flight controller

ARK FMU V6x

### Vehicle type

Fixed Wing

benjinne, 2023-08-04:
> What version of QGC are you on? Please make sure it's the latest and report back

4.2.8

junwoo091400, 2023-08-04:
@benjinne once the Linux Release of the CI is done in this PR: https://github.com/mavlink/qgroundcontrol/pull/10759, could you test it out to see if it fixed it?

benjinne, 2023-08-04:
> @benjinne once the Linux Release of the CI is done in this PR: [mavlink/qgroundcontrol#10759]

I can confirm this fixes my issue

[image]

benjinne, 2023-08-04:
Hower, the channel monitor is still blank even when I connect my RC controller

benjinne, 2023-08-04:
channel monitor shows up once my RC controller is connected
[image]

closed 2023-08-05 by DonLakeFlyer