PX4/PX4-Autopilot #21932: in  docker , the  gazebo cannot display correctly

XXLiu-HNU, 2023-08-07:
### Describe the bug

When I compile the program inside the docker container, the gazebo graphical interface cannot be displayed correctly. I already use xhost + .

`[Msg] Waiting for master.
INFO  [commander] LED: open /dev/led0 failed (22)
WARN  [health_and_arming_checks] Preflight Fail: ekf2 missing data
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 172.18.0.5
[Err] [RenderEngine.cc:749] Can't open display: :0
[Wrn] [RenderEngine.cc:89] Unable to create X window. Rendering will be disabled
[Wrn] [RenderEngine.cc:292] Cannot initialize render engine since render path type is NONE. Ignore this warning ifrendering has been turned off on purpose.
[Wrn] [GuiIface.cc:120] could not connect to display :0
[Msg] Could not load the Qt platform plugin "xcb" in "" even though it was found.
[Err] [GuiIface.cc:124] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.`

### To Reproduce

`xhost +
docker exec -it uav1 bash
cd src/px4
make px4_sitl gazebo-classic

`

### Expected behavior

gazebo can display normally

### Flight Log

none

### Software Version

v1.14

### Flight controller

none

### Vehicle type

None

biagio-trimarchi, 2023-08-07:
@XXLiu-HNU I had a similar problem some time ago.
What worked for me was to set the DISPLAY environmental variable to 0.

If you want to try, in the terminal of your container run
```
export DISPLAY=:0
```

If it works, you add that line to the `.bashrc` file of your container so that is set automatically when you run the container. To do this you can run 
```
echo "source DISPLAY=:0" >> ~/.bashrc
```

XXLiu-HNU, 2023-08-09:
Thank you for your answer, but the bug still exists

XXLiu-HNU, 2023-08-09:
[image]

LiuWhale, 2023-10-17:
I have the same issue, but my env is ubuntu22.04 and I use the humble container built by myself.
[image]

Hefero, 2023-11-19:
I was able to run PX4-Autopilot/Mavlink separatedly from gazebo, under docker via vnc,
doing something like this:

cd <PX4-Autopilot_clone>
source Tools/simulation/gazebo-classic/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/simulation/gazebo-classic/sitl_gazebo-classic

Terminal1:
sudo HEADLESS=1 make px4_sitl_default gazebo-classic
Terminal2:
roslaunch mavros px4.launch fcu_url:="udp://:14540@127.0.0.1:14557"
Terminal3:
roslaunch gazebo_ros empty_world.launch world_name:=$(pwd)/Tools/simulation/gazebo-classic/sitl_gazebo-classic/worlds/empty.world
[image]