PX4/PX4-Autopilot #21952: [Bug] motors spin after arming with 0 throttle

christianrauch, 2023-08-09:
### Describe the bug

With the new Control Allocation in place of the olf Mixing, I am encountering the following behaviour: After arming with sticks (throttle down, yaw right) the motors immediately start spinning although the throttle is still 0.

This is a regression from `stable`, where motors would only spin when the throttle is increased.

### To Reproduce

Use the `px4_raspberrypi_default` target with this config:
````sh
#!/bin/sh
# PX4 commands need the 'px4-' prefix in bash.
# (px4-alias.sh is expected to be in the PATH)
. px4-alias.sh

param select eeprom/parameters
param import

# system_power not implemented
param set CBRK_SUPPLY_CHK 894281

param set-default BAT1_V_DIV 5.7

param set SYS_AUTOCONFIG 0
param set MAV_TYPE 2
param set SYS_AUTOSTART 4001

param set-default PCA9685_RATE 50

param set COM_ARM_SDCARD 0

dataman start

load_mon start

battery_status start

# start logging
logger start -f

rc_input start -d /dev/ttyAMA0

rc_update start
sensors start
navigator start
ekf2 start
land_detector start multicopter
mc_hover_thrust_estimator start
flight_mode_manager start
mc_pos_control start
mc_att_control start
mc_rate_control start

param set-default SYS_HAS_BARO 0
param set-default SYS_HAS_GPS 0
param set-default COM_RC_IN_MODE 0 # RC Transmitter only
# param set-default COM_RC_IN_MODE 1 # Joystick only

param set-default COM_RCL_EXCEPT 4

param set-default COM_FLTMODE1 0
param set-default COM_FLTMODE2 1
param set-default COM_FLTMODE3 2
param set-default COM_FLTMODE4 3
param set-default COM_FLTMODE5 4
param set-default COM_FLTMODE6 5

param set-default SDLOG_MODE 2 # from boot until shutdown

param set-default CA_ROTOR_COUNT 4

param set-default CA_ROTOR1_PX +0.08
param set-default CA_ROTOR3_PX +0.08
param set-default CA_ROTOR0_PX -0.08
param set-default CA_ROTOR2_PX -0.08

param set-default CA_ROTOR3_PY -0.1025
param set-default CA_ROTOR2_PY -0.1025
param set-default CA_ROTOR1_PY +0.1025
param set-default CA_ROTOR0_PY +0.1025

# map input channels
param set RC_MAP_PITCH 3
param set RC_MAP_ROLL 2
param set RC_MAP_THROTTLE 1
param set RC_MAP_YAW 4
param set RC_MAP_FLTMODE 6

# Use a positive value for a rotor with CCW rotation. Use a negative value for a rotor with CW rotation.
param set CA_ROTOR0_KM -0.05
param set CA_ROTOR1_KM +0.05
param set CA_ROTOR2_KM +0.05
param set CA_ROTOR3_KM -0.05

# map channels : motors 1:1
param set PCA9685_FUNC1 101
param set PCA9685_FUNC2 102
param set PCA9685_FUNC3 103
param set PCA9685_FUNC4 104

param set PWM_TEST_FUNC1 101
param set PWM_TEST_FUNC2 102
param set PWM_TEST_FUNC3 103
param set PWM_TEST_FUNC4 104

mavlink start -x -u 14556 -r 1000000 -p

echo "mpu9250 >>>"
mpu9250_i2c start -I #-a 0x68
sleep 1
ak8963 start -I -a 0x0C
sleep 1
echo "mpu9250 ###"

echo "pwm >>>"
pca9685_pwm_out start
control_allocator start
echo "pwm ###"

attitude_estimator_q start

manual_control start

commander start

mavlink boot_complete
````

If it helps, there are the parameters: [param.txt](https://github.com/PX4/PX4-Autopilot/files/12302380/param.txt) (exported via `./bin/px4-param show > param.txt`).

I am using `pca9685_pwm_out` to send PWM signals to the ESCs and `rc_input` with SBUS to get RC input.

### Expected behavior

Motors should only spin when giving thrust.

### Screenshot / Media

In this screenshot, you can see that the sticks are in the arming position (throttle -1, yaw +1) but two of the four motors already get PWM signals above 1000us which makes them spin.
[image]

### Flight Log

QGroundControl is unable to sort logs properly so here is a log from a different incident but with the same behaviour: https://logs.px4.io/plot_app?log=81646436-ee4f-4c24-b9ef-91b2a2905ea9.

### Software Version

```
pxh> ver all
HW arch: PX4_RASPBERRYPI
PX4 git-hash: c8b66dc40c9ee53e3ee06c7953b9a8c0e0e38739
PX4 version: 1.14.0 0 (17694720)
PX4 git-branch: rpi
OS: Linux
OS version: Release 6.1.21 (100734463)
Build datetime: Aug  9 2023 10:44:42
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 10.2.1 20210110
PX4GUID: 100449495052303030443030303030303030
UNKNOWN MCU
```

### Flight controller

Raspberry Pi 2

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

UART and I2C

danielhonies, 2023-08-09:
For multirotors the default behaviour is for the motors to spin at minimum speed once the drone is armed. It is also how every other multirotor platform (DJI etc) works. IMO I would argue them not spinning is even more unsafe as then it is not apparent if the drone is armed or not.

christianrauch, 2023-08-09:
> For multirotors the default behaviour is for the motors to spin at minimum speed once the drone is armed.

This wasn't the case for `stable`. More hobbyistic firmwares, such as betaflight, will arm without spinning the motors.

> IMO I would argue them not spinning is even more unsafe as then it is not apparent if the drone is armed or not.

I don't see why after all these years it is now deemed to be more safe to have the motors spin at 0 throttle. Usually arming is indicated by an LED colour or sound. Also, as you can see from the readings above, the motors do not just spin equally at low speeds, but they indeed spin based on the given yaw. This does not make sense to me. Why should the FC account for the given yaw rate, if there is no throttle?

Is this explicit change in behaviour, forcing motors to spin right after arming with no throttle, discussed and documented somewhere?

danielhonies, 2023-08-09:
I don't know how your stable or ESCs were configured before but I have used PX4 for 5 years with dozens of vehicles and every single one of them was by default spinning the motors after arm. As discussed in the PR the minimum PWM was increased so that even those ESCs that wouldn't turn, would now turn. The intended behaviour has always been for the motors to spin and for most ESCs that has always been the case. 

One of the reasons is that a drone is expected to hold it's attitude even if throttle is put to zero while mid-air.

danielhonies, 2023-08-09:
Why minimum thrust is necessary is documented here: 
https://docs.px4.io/main/en/config_mc/pid_tuning_guide_multicopter.html#airmode-mixer-saturation

christianrauch, 2023-08-09:
> I don't know how your stable or ESCs were configured before but I have used PX4 for 5 years with dozens of vehicles and every single one of them was by default spinning the motors after arm.

Ok. At least in betaflight this seems to be [configurable via `MOTOR_STOP`](https://betaflight.com/docs/development/Controls). I guess I got so used to the feature and found the same behaviour in the `stable` branch that I assumed this is expected. I personally feel safer if the motors do not spin until throttle is applied and it would be great if a similar option would exist in PX4.

> One of the reasons is that a drone is expected to hold it's attitude even if throttle is put to zero while mid-air.

During flight, this behaviour makes sense of course. But I don't see why the FC should adjust for the yaw when throttle has never been applied since arming. The motors could just equally spin when armed and only adjust for the control inputs once throttle is applied for the first time since arming.

danielhonies, 2023-08-09:
If you want the behaviour to be like before, you can adjust the minimum PWM parameters.

junwoo091400, 2023-08-11:
> I have used PX4 for 5 years with dozens of vehicles and every single one of them was by default spinning the motors after arm

I had the same experience, but @christianrauch you are saying that stable release (v1.13.3) does not spin the motors when throttle is at 0 after arming right?

christianrauch, 2023-08-14:
> I had the same experience, but @christianrauch you are saying that stable release (v1.13.3) does not spin the motors when throttle is at 0 after arming right?

Yes. I could arm without the motors spinning and they would only spin once I increase the throttle.

junwoo091400, 2023-08-15:
> I could arm without the motors spinning and they would only spin once I increase the throttle.

Have you checked out the v1.14 Release Note? It could be related to the change of 900 to 1000 for disarmed motor PWM value: https://docs.px4.io/main/en/releases/1.14.html#upgrade-guide

@MaEtUgR any chance that that change changes the ESC calibration step, and hence causes motor to spin after arming (whereas in v1.13 setup it wouldn't have)?

julianoes, 2023-10-19:
This has beem answered and there was no more follow up, closing.

closed 2023-10-19 by julianoes

MaEtUgR, 2023-11-16:
@christianrauch Just to clarify: For any multirotor the motors have to all spin upon arming even with the minimum commanded thrust. By design the defaults were changed to reach that behavior with more available hardware here: https://github.com/PX4/PX4-Autopilot/pull/21513 It's part of why going through this actuator range configuration step 7 here i so important: https://docs.px4.io/main/en/advanced_config/esc_calibration.html#actuatorconfig_step is so important.

Technical reason: If the motors don't spin with minimal commanded thrust then the ESCs will stop the motors also in air whenever they get a minimum thrust command. ESC cannot sync and spin up motors that were stopped quick enough again which leads to big oscillations amplifying the problem further and eventually a crash. When hovering and never hitting the lower actuator saturation everything might look fine but once that state is reached in air a crash is often unavoidable.

mengchaoheng, 2024-12-05:
@MaEtUgR @christianrauch Now in v1.15.1, after setting arm by default, the motor of the fixed-wing aircraft does not rotate. Is there any way to make it rotate at a low speed? I need to know whether it has been unlocked.