PX4/PX4-Autopilot #21953: [Bug] SITL ignoring virtual Joystick as manual input

hamishwillee, 2023-08-09:
### Describe the bug

Virtual joysticks are supposed to be equivalent to a real joystick, so if you enable them the vehicle should pass the same preflight checks that it would pass with a real joystick. However this is not the case - enabling virtual joysticks does not pass the "No manual control input".

However plugging in a joystick does disable that check. It also latches the result, so if you disconnect the joystick the arming warning does not appear, and you can take off with virtual joysticks. If you then restart QGC it reverts back to not allowing arming.

### To Reproduce

1. Connect QGC to SITL and disconnect any joystick or RC controller
2. Select QGC Arming UI (select "Not Ready") in Flight view. Should display reasons not to fly.
3. Initially 
   [image]
4. Set COM_RC_IN_MODE param to "Joystick" (or some other Joystick mode).
5. Enable virtual joysticks in application settings
   [image]
   Note that joysticks appear on screen but the Preflight warning does not.
 6. Plug in a joystick. Note that the warning disappears.
 7. Remove joystick. Note that the warning does not reappear.
 8. Use virtual joysticks to take off.
 

### Expected behavior

If I enable the virtual joystick a joystick should be detected by PX4.

### Flight Log

Doubt it will be useful, but: https://logs.px4.io/plot_app?log=286c6126-b2e4-4541-a782-18c16f700d2c

This shows initial state - not passing check even though had virtual joysticks, then connecting real joystick and preflight check passing.

### Software Version

QGC daily from 20230810 (as shown) on Windows 11, connected to Headless PX4 SITL Gazebo main (also 20230810 - yesterday) running on WSL2 (Ubuntu 22.04).

[image]

### Flight controller

Simulator

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

NA

bkueng, 2023-08-10:
Looks like it's a QGC bug, so tracking it there: https://github.com/mavlink/qgroundcontrol/issues/10664

closed 2023-08-10 by bkueng

MaEtUgR, 2023-08-15:
> Looks like it's a QGC bug

I can confirm. When using an old QGC version it still works as expected.

MaEtUgR, 2023-08-15:
Also the latest one works thanks to https://github.com/mavlink/qgroundcontrol/pull/10758