PX4/PX4-Autopilot #21964: [Bug] Offboard control not available even when position control is functioning

Brendan-D-Kearney, 2023-08-14:
### Describe the bug

The FCU rejects offboard mode with the message `check for valid position estimate` in QGC. All other flight modes including position control work perfectly, suggesting there is a valid position estimate available. 

### To Reproduce

1. Use MoCap for external position estimate
2. Spoof GPS to give a home position lock if in GPS denied environment
3. Send offboard local position commands through MAVROS (i.e. standard offboard example)
4. Attempt to switch to offboard control after all checks have been made

### Expected behavior

Vehicle enters offboard control and moves to position setpoint.

### Flight Log

Offboard control error does not show up, despite QGC rejection messages, and MAVROS console rejection messages: https://review.px4.io/plot_app?log=519ccd46-cda9-4e18-b68f-82bc658b2206

Flight Behaviour Description: 
1. Arm the drone
2. Attempt to initiate offboard mode
3. offboard mode rejected - switch to position mode
4. illustrate working position mode flight

### Software Version

1.13.3

### Flight controller

pixhawk 6c

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

Default setup: https://holybro.com/products/px4-vision-dev-kit-v1-5

### Additional context

Offboard control is rejected at all stages, whether armed or disarmed, or in flight. To publish setpoints we are using this simple [offboard control example](https://docs.px4.io/v1.12/en/ros/mavros_offboard.html). Included below are the outputs of MAVROS and the offboard control setpoint publishing message. 

This setup is indoors and uses MoCap (vicon) in a completely denied GPS environment. We have followed [this guide](https://docs.px4.io/main/en/ros/external_position_estimation.html) religiously to get position control working through MAVROS. Despite position control working we have no offboard control.

We are continuing to actively troubleshoot.

Setpoint Message: 
```
header: 
  seq: 4902
  stamp: 
    secs: 1692035105
    nsecs:    764315
  frame_id: "\x01"
pose: 
  position: 
    x: 0.0
    y: 0.0
    z: 0.5
  orientation: 
    x: 0.0
    y: 0.0
    z: 0.0
    w: 0.0
```

MAVROS terminal:
[... 344 lines cut ...]
[ INFO] [1692035053.633685229]: VER: 1.1: Board hardware:      00000038
[ INFO] [1692035053.638221191]: VER: 1.1: VID/PID:             3185:0038
[ INFO] [1692035053.638770391]: VER: 1.1: UID:                 3430510d38303334
[ WARN] [1692035053.640363791]: CMD: Unexpected command 520, result 0
[ INFO] [1692035067.612355008]: GF: mission received
[ INFO] [1692035067.614542708]: RP: mission received
[ INFO] [1692035067.618215095]: WP: mission received
[ INFO] [1692035113.549895314]: FCU: Armed by RC
[ INFO] [1692035113.602404735]: FCU: [logger] /fs/microsd/log/2023-08-14/17_45_13.ulg
[ERROR] [1692035116.642645573]: FCU: Switching to Offboard is currently not available
[ INFO] [1692035118.851945007]: FCU: Takeoff detected
[ WARN] [1692035171.534814115]: CMD: Unexpected command 176, result 0
[ INFO] [1692035177.985880662]: FCU: Landing detected
[ INFO] [1692035178.499885066]: FCU: Disarmed by landing
[ WARN] [1692035178.509980413]: PR: Param CAL_ACC0_ID (32/854): <value><i4>4259850</i4></value> different index: 32/856
[ WARN] [1692035178.510666587]: PR: Param CAL_ACC0_PRIO (33/854): <value><i4>50</i4></value> different index: 33/856
[ WARN] [1692035178.511317301]: PR: Param CAL_ACC0_ROT (34/854): <value><i4>-1</i4></value> different index: 34/856
[ WARN] [1692035178.511945392]: PR: Param CAL_ACC0_XOFF (35/854): <value><double>-0.22992914915084839</double></value> different index: 35/856
[ WARN] [1692035178.512614992]: PR: Param CAL_ACC0_XSCALE (36/854): <value><double>0.98118484020233154</double></value> different index: 36/856
[ WARN] [1692035178.513582007]: PR: Param CAL_ACC0_YOFF (37/854): <value><double>-0.068059690296649933</double></value> different index: 37/856
[ WARN] [1692035178.514317353]: PR: Param CAL_ACC0_YSCALE (38/854): <value><double>0.98943412303924561</double></value> different index: 38/856
[ WARN] [1692035178.515119330]: PR: Param CAL_ACC0_ZOFF (39/854): <value><double>-0.022407421842217445</double></value> different index: 39/856
[ WARN] [1692035178.515814516]: PR: Param CAL_ACC0_ZSCALE (40/854): <value><double>0.99668693542480469</double></value> different index: 40/856
[ INFO] [1692035184.312262052]: FCU: Kill-switch engaged
```

Brendan-D-Kearney, 2023-08-14:
> This is lilely what is breaking your position estimate, since the gps is reporting that the vehicle is not moving

When we look at the MAVLink messages it indicates that GPS position does update, as the MAVROS fake_gps package passes the vicon data and transforms it into a gps frame. We have disabled all GPS checks and fusion however, so there should be no reason GPS is causing an issue. We were unable to get position control to work at all without spoofing a gps lock at least for the initial flight. 

But previously while position control was **not** working correctly we were able to enter offboard mode, and as you could guess the drone flew erratically due to it. Below is the PX4 parameters we are using:

```
# Onboard parameters for Vehicle 1
#
# Stack: PX4 Pro
# Vehicle: Multi-Rotor
# Version: 1.13.2 
# Git Revision: 46a12a09bf000000
#
# Vehicle-Id Component-Id Name Value Type
1	1	ASPD_SCALE_1	1.000000000000000000	9
1	1	BAT1_A_PER_V	36.363998413085937500	9
1	1	BAT1_CAPACITY	-1.000000000000000000	9
1	1	BAT1_I_CHANNEL	-1	6
1	1	BAT1_N_CELLS	3	6
1	1	BAT1_R_INTERNAL	-1.000000000000000000	9
1	1	BAT1_SOURCE	0	6
1	1	BAT1_V_CHANNEL	-1	6
1	1	BAT1_V_CHARGED	4.050000190734863281	9
1	1	BAT1_V_DIV	18.181999206542968750	9
1	1	BAT1_V_EMPTY	3.500000000000000000	9
1	1	BAT1_V_LOAD_DROP	0.300000011920928955	9
1	1	BAT2_A_PER_V	36.367515563964843750	9
1	1	BAT2_CAPACITY	-1.000000000000000000	9
1	1	BAT2_I_CHANNEL	-1	6
1	1	BAT2_N_CELLS	0	6
1	1	BAT2_R_INTERNAL	-1.000000000000000000	9
1	1	BAT2_SOURCE	-1	6
1	1	BAT2_V_CHANNEL	-1	6
1	1	BAT2_V_CHARGED	4.050000190734863281	9
1	1	BAT2_V_DIV	18.100000381469726563	9
1	1	BAT2_V_EMPTY	3.500000000000000000	9
1	1	BAT2_V_LOAD_DROP	0.300000011920928955	9
1	1	BAT_AVRG_CURRENT	15.000000000000000000	9
1	1	BAT_CRIT_THR	0.070000000298023224	9
1	1	BAT_EMERGEN_THR	0.050000000745058060	9
1	1	BAT_LOW_THR	0.150000005960464478	9
1	1	BAT_N_CELLS	3	6
1	1	BAT_V_CHARGED	4.050000190734863281	9
1	1	BAT_V_EMPTY	3.500000000000000000	9
1	1	BAT_V_LOAD_DROP	0.300000011920928955	9
1	1	BAT_V_OFFS_CURR	0.000000000000000000	9
1	1	CAL_ACC0_ID	4259850	6
1	1	CAL_ACC0_PRIO	50	6
1	1	CAL_ACC0_ROT	-1	6
1	1	CAL_ACC0_XOFF	-0.011063143610954285	9
1	1	CAL_ACC0_XSCALE	0.980389595031738281	9
1	1	CAL_ACC0_YOFF	0.111430212855339050	9
1	1	CAL_ACC0_YSCALE	0.981184959411621094	9
1	1	CAL_ACC0_ZOFF	0.028970301151275635	9
1	1	CAL_ACC0_ZSCALE	0.990540683269500732	9
1	1	CAL_ACC1_ID	2490378	6
1	1	CAL_ACC1_PRIO	50	6
1	1	CAL_ACC1_ROT	-1	6
1	1	CAL_ACC1_XOFF	0.185475841164588928	9
1	1	CAL_ACC1_XSCALE	1.000000000000000000	9
1	1	CAL_ACC1_YOFF	0.319903492927551270	9
1	1	CAL_ACC1_YSCALE	1.000000000000000000	9
1	1	CAL_ACC1_ZOFF	-0.091852337121963501	9
1	1	CAL_ACC1_ZSCALE	1.000000000000000000	9
1	1	CAL_ACC2_ID	0	6
1	1	CAL_ACC3_ID	0	6
1	1	CAL_AIR_CMODEL	0	6
1	1	CAL_AIR_TUBED_MM	1.500000000000000000	9
1	1	CAL_AIR_TUBELEN	0.200000002980232239	9
[... 777 lines cut ...]
1	1	UAVCAN_ESC_IDLT	1	6
1	1	UAVCAN_LGT_ANTCL	2	6
1	1	UAVCAN_LGT_LAND	0	6
1	1	UAVCAN_LGT_NAV	3	6
1	1	UAVCAN_LGT_STROB	1	6
1	1	UAVCAN_NODE_ID	1	6
1	1	UAVCAN_PUB_MBD	0	6
1	1	UAVCAN_PUB_RTCM	0	6
1	1	UAVCAN_SUB_ASPD	0	6
1	1	UAVCAN_SUB_BARO	0	6
1	1	UAVCAN_SUB_BAT	0	6
1	1	UAVCAN_SUB_BTN	0	6
1	1	UAVCAN_SUB_DPRES	0	6
1	1	UAVCAN_SUB_FLOW	0	6
1	1	UAVCAN_SUB_GPS	1	6
1	1	UAVCAN_SUB_HYGRO	0	6
1	1	UAVCAN_SUB_ICE	0	6
1	1	UAVCAN_SUB_IMU	0	6
1	1	UAVCAN_SUB_MAG	1	6
1	1	UAVCAN_SUB_RNG	0	6
1	1	VTO_LOITER_ALT	80.000000000000000000	9
1	1	VT_B_DEC_MSS	2.000000000000000000	9
1	1	VT_B_REV_DEL	0.000000000000000000	9
1	1	WV_EN	0	6
```

Jaeyoung-Lim, 2023-08-14:
> When we look at the MAVLink messages it indicates that GPS position does update, as the MAVROS fake_gps package passes the vicon data and transforms it into a gps frame. We have disabled all GPS [...]

This is actually a good hint that your external vision input might have wrong frame definitions. The external pose estimate works without GPS spoofing.

Brendan-D-Kearney, 2023-08-14:
I believe the external vision estimate is correct, but there is some underlying issues with the current version of PX4 and MAVROS that do not correctly account for complete lack of GPS. Without a GPS fix, the quad has no designated home position, but we cannot manually set a home position without faking the GPS. The external position estimate guide that we have followed states that once everything is correct, you should get a green light on the pixhawk and should be able to fly with position control active. We can only achieve both of these things by running fake_gps, an accurate external position estimate does not seem to be enough for the pixhawk.

Brendan-D-Kearney, 2023-08-14:
I was able to fix the issue by editting the MAVROS px4_config.yaml file. It appears if you have:
```
Setpoint Position
listen:
tf: true
```
then PX4 will not allow you to switch to offboard mode. Perhaps this can be updated in the future so as to tell users why offboard mode is not being enabled.

closed 2023-08-14 by Brendan-D-Kearney

Jaeyoung-Lim, 2023-08-14:
> but there is some underlying issues with the current version of PX4 and MAVROS that do not correctly account for complete lack of GPS. Without a GPS fix, the quad has no designated home position, but we cannot manually set a home position without faking the GPS. 

@Brendan-D-Kearney I am not sure if this is true. Are you sure you have configured the EKF correctly to ignore GPS?

Brendan-D-Kearney, 2023-08-15:
@Jaeyoung-Lim You can view this issue: https://github.com/PX4/PX4-Autopilot/issues/21939 which is posted by a fellow in the lab using the same setup. This issue is from before using any spoofed GPS, and you can see we are unable to enter position control and thus have never attempted offboard control. We have turned off all available settings for GPS in the EKF tunings available with QGC. Even when troubleshooting this issue when I lose the spoofed GPS signal and turn off all GPS parameters I lose all position control capability, despite the fact that position control _clearly_ works when GPS is spoofed.

Jaeyoung-Lim, 2023-08-15:
@Brendan-D-Kearney I understand what is happening on your system. What I am trying to say is that it is likely something particular for your platform. If this is not the case, it should be considered as a bug. However, the fact that you are using v1.13.2, I am no longer sure what is on that release.

enord-co, 2023-08-30:
@Jaeyoung-Lim I hope you are doing well.
I have a problem I have pixhawk 6x as soon as i am changing **SYS_MC_EST_GROUP** to **Q attitude estimator (no position)** i am not getting /mavros/imu/data (1.13.0/1.13.1/1.13.3) but as i downgrade the firmware version to 1.12 and so on it is publishing the data.