PX4/PX4-Autopilot #21965: [Bug] Inquiry on Resolving Continuous Reboot Issue with Custom Flight Control Board

yukaichao, 2023-08-15:
### Describe the bug

Dear PX4 Developers,

I am deeply appreciative of your contributions, which have significantly benefited my project. I have encountered some issues while developing a custom flight control board, and I am hoping you could provide some guidance.

In my design, I referred to the PX4FMUv2.4.6.sch in the PIXHAWK hardware design and made some hardware changes. The original STM32F427VIT6 main control was replaced with STM32F427IIH6, and the STM32F100 was replaced with STM32F100R8H6B. After these changes, I found that the flight control board keeps rebooting after the PX4 firmware starts.

I have checked the hardware and firmware, and even uploaded the APM firmware to confirm there are no hardware issues. However, I have not been able to identify the key source of the continuous reboot issue with the PX4 firmware. I would like to ask for your advice on what could possibly be causing the continuous reboots? Do you have any recommended solutions, or do you need more information from me?

I am immensely grateful for any help you can provide! I look forward to your reply.

Best regards

### To Reproduce

homemade flight control board

### Expected behavior

It should boot up normally, and it can be connected to the ground station using USB.

### Flight Log

No Logs Available

### Software Version

v1.13.3

### Flight controller

homemade flight control board

### Vehicle type

None

ShiauweiZhao, 2023-08-15:
do you have hardfault log？

yukaichao, 2023-08-15:
> 你有硬故障日志吗？
这个日志怎么获取

yukaichao, 2023-08-15:
> do you have hardfault log？
Hello, to access this log, you can refer to the documentation. After flashing the official compiled firmware, the USB keeps disconnecting and reconnecting (the USB device name is Legacy FMU).

closed 2023-09-05 by yukaichao