PX4/PX4-Autopilot #21966: Support for DO_SET_ACUTUATOR and DO_SET_SERVO MAVLink commands

junwoo091400, 2023-08-15:
### Describe problem solved by the proposed feature

I have noted that a lot of users (e.g. https://discuss.px4.io/t/using-do-set-actuator-to-releasing-the-payload/28686) were asking for ways to control the servo, or an actuator via the DO_SET_ACTUATOR/SERVO messages coming from an external entity (e.g. onboard computer).

This definitely sounds like a basic feature that we should support, as it can be used for all kinds of applications that controls *any sort of actuator onboard the drone,

As we now have the `Actuators` and `Servos` concept in place with control allocation, it should be pretty straightforward to implement, I think.

### Describe your preferred solution

Add the logic in `mavlink_receiver.cpp` like was done in v1.13:

https://github.com/PX4/PX4-Autopilot/blob/1c8ab2a0d7db2d14a6f320ebd8766b5ffaea28fa/src/modules/mavlink/mavlink_receiver.cpp#L554-L586

### Describe possible alternatives

None

### Additional context

I have posted a bit of history behind the support for these commands here: https://discuss.px4.io/t/using-do-set-actuator-to-releasing-the-payload/28686/3, however, to summarize:

1. This feature was supported in v1.13 ([related comment](https://github.com/PX4/PX4-Autopilot/pull/18265#issuecomment-923681920)), and was utilizing control groups convention (which was removed after complete switch to control allocation in v1.14)
2. It was then removed when purging old mixer system (using control groups, etc): https://github.com/PX4/PX4-Autopilot/commit/13f9eabd70de2038330e89ab46d5cb3f1cde38ec

Therefore, I would say this was an accidental deletion and the feature should be put back in place.

hamishwillee, 2023-08-16:
@junwoo091400 Whaaaaaattttt? If this was removed it should be put back, or the docs need to be cleaned up- the most relevant section is https://docs.px4.io/main/en/payloads/#generic-actuator-control which states that they are supported and can be set using the actuator setup screen.

Can you contact me directly once some decision is made - I would say this is a critical drop of functionality, since otherwise you're pretty much limited to your gripper payload stuff for non-camera payloads.

hamishwillee, 2023-09-20:
@mrpollo IMO this is a critical defect against PX4 in general and PX4 v1.14 in particular.

hamishwillee, 2023-09-20:
I've renamed to highlight that this needs some eyes on it.

hamishwillee, 2023-11-01:
@junwoo091400 FYI was removed by @dagar in https://github.com/PX4/PX4-Autopilot/commit/13f9eabd70de2038330e89ab46d5cb3f1cde38ec along with the underlying support for actuator groups. There does still appear to be some handling in missions though I haven't tested that.

To get this back we'd need a new implementation as the underlying foundations of the original implementation no longer exist.

julianoes, 2023-11-14:
Actually, are we talking about the command [DO_SET_ACTUATOR](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_ACTUATOR) or the offboard message [SET_ACTUATOR_CONTROL_TARGET](https://mavlink.io/en/messages/common.html#SET_ACTUATOR_CONTROL_TARGET)? I believe the former is still available but the latter has been removed.

hamishwillee, 2024-01-31:
@junwoo091400 Yes, we're talking about [DO_SET_ACTUATOR](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_ACTUATOR)

No, it was removed in https://github.com/PX4/PX4-Autopilot/commit/13f9eabd70de2038330e89ab46d5cb3f1cde38ec - probably by accident, but hard to know - see https://github.com/PX4/PX4-Autopilot/commit/13f9eabd70de2038330e89ab46d5cb3f1cde38ec#diff-22f2d2347fbee4247ef71142274d558334f8991fb89dca50f7505ba05bf9873f

I find this incredibly frustrating. Yet another request for it here: https://github.com/PX4/PX4-user_guide/issues/3028

FYI @mrpollo - is it time for me to remove all the instructions on how to use this essential feature from the docs or can we get this restored?

bkueng, 2024-02-01:
@hamishwillee I just tested `DO_SET_ACTUATOR` on 1.14 via MAVLink command and it works fine. The code you reference did some special handling for _actuator_controls_pubs (now removed).
Therefore I'm closing this.

closed 2024-02-01 by bkueng

bkueng, 2024-02-01:
There was just no ACK sent back. Maybe that caused issues with some MAVLink clients. I'm adding it in https://github.com/PX4/PX4-Autopilot/pull/22711

mrpollo, 2024-02-01:
Hey @bkueng do we need to make any updates to docs?

hamishwillee, 2024-02-01:
> There was just no ACK sent back. Maybe that caused issues with some MAVLink clients. 

Thanks very much @bkueng - if you say tested and works more than good enough for me.

FWIW I do rely on acks to test some of this stuff, but did not test this. Just inspected code and didn't see the handler for this command

bkueng, 2024-02-02:
Let's add it to 1.14 as well then: https://github.com/PX4/PX4-Autopilot/pull/22715

> Hey @bkueng do we need to make any updates to docs?

We could explicitly mention `COM_PREARM_MODE` in the first note under https://docs.px4.io/main/en/payloads/.

BobbyGeanie, 2026-10-14:
Faced this in 2026, very annoying. My big guess is that `MAV_CMD_DO_SET_SERVO` is deprecated, I don't understand why they don't mentioned on [this](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_SERVO). The work around is use `MAV_CMD_DO_SET_ACTUATOR` instead, and you map servo in QGC actuator configuration instead. Hope this help to next person facing this issue!

BobbyGeanie, 2026-10-20:
@julianoes Hi, I agree, because I can compile code with Mavlink lib, but error thrown when connect to PX4. However, if this is the case, it was deprecated very long time ago **at least from px4 v1.13**. I didn’t check the default px4 which come with my **HolyBro 6c mini** but I think it is px4 v1.13, since it didn’t work, I thought it’s too old at first, so I flash v1.17 and still don’t work, after that then I go with workaround. Hope this help, thank you.

hamishwillee, 2026-10-21:
> 29\. [@BobbyGeanie](https://github.com/BobbyGeanie) my guess is that the "deprecation" is only on the PX4 side but not the MAVLink side. Can you confirm [@hamishwillee](https://github.com/hamishwillee)? And I wonder whether we should/could just support both anyway in PX4.

It isn't deprecated in MAVLink https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_SERVO.

Though arguably it is superseded and should eventually be deprecated. Reason being it is better. There isn't anything you can do wth set_servo that set-actuator doesn't do more easily.