PX4/PX4-Autopilot #21974: [Bug] Gazebo's drone is not working in an external HITL

ymh1995s, 2023-08-17:
### Describe the bug

Hello.
I want to simulate it on Gazebo in conjunction with the actual drone that removed the propeller.
The reason for using a real drone is to check with a simulator how the pixhawk 6c controls the drone when receiving spoofing GPS signals.

I found the document “HITL Simulation” and I followed the procedure
When the take-off order was issued, Gazebo’s drone did not move despite checking the “in flight” letters in QGroundControl and the real drone’s motor rotation.

### To Reproduce

1. Ubuntu 20.04 install

2. latest PX Autopilot install
2-1. git clon [GitHub - PX4/PX4-Autopilot: PX4 Autopilot Software](https://github.com/PX4/PX4-Autopilot.git) --recursive
2-2. bash ./Tools/setup/ubuntu.sh
2-3. DON_T RUN=1 make px4_sitl gazebo
2-4. changed “serialEnabled=1”, “hil_mode=1” in iris.sdf
2-5. source Tools/simulation/gazebo-classic/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default

3. latest QGroundControl install
3-1. changed “external HITL” in Hardware in the loop simulation Menu.
3-2. Only checked one, UDP auto connect
3-3. I chose a drone Quadroter generic
3-4. Enable virtual joystick.

4. After the installation, I closed all the programs and ran them again.
4-1. gazebo Tools/simulation/gazebo-classic/sitl_gazebo_classic/world/hitl_iris.world
4-2. I was able to confirm that it was connected to tty/ACM0 (my drone, 6c).
4-3. QGroundControl executed.
4-4. I checked the HITL activation message in the QGroundControl pop-up and confirmed that the instrument cluster and head move when the real drone is moved by hand.

5. We ordered take-off via QGroundControl, and there was a problem
5-1. Checked “in flight” green text on QGroundControl.
5-2. If the battery was connected, the rotor of the real drone also rotated.
5-3. Gazebo’s drone doesn’t move(rotor does not rotate).

### Expected behavior

real Drone, gazebo drone and QGroundControl must all operate.
The gazebo drone takes off under take-off orders.
(Probably, the real drone does not increase in altitude, so gazebo drone rises to infinity.)

### Screenshot / Media

[image]

### Flight Log

https://logs.px4.io/plot_app?log=35719951-d656-4938-9ed3-b431d7d19cb4

### Software Version

v1.13.3

### Flight controller

pixhawk 6c

### Vehicle type

None

NanyongGit, 2024-03-12:
+1 facing the same issue at the moment

ifiksel1, 2024-08-05:
I followed all your steps to recreate the issue using a Cube Orange FC, The drone flies in gazebo and it shows in QGC but the motor on the vehicle doesn't move. When I move the FC by hand, the instrument cluster and head doesn't move. Has this been solved?