PX4/PX4-Autopilot #21978: Cloned it, first-time compilation failed.

swj735245824, 2023-08-18:
### Describe the bug

When I clone the project and execute 'make px4_sitl jmavsim' to compile the source code for the first time, the console compilation fails with the following error: 
> PX4-Autopilot/src/drivers/gps/devices/src/mtk.cpp:220:18: fatal error: no member named 'latitude_deg' in 'sensor_gps_s'
> _gps_position->latitude_deg = packet.latitude / 1e6; // from degrees*1e6 to degrees
> …
> PX4-Autopilot1/src/drivers/gps/devices/src/ubx.cpp:1898:19: fatal error: no member named 'latitude_deg' in 'sensor_gps_s'
>                         _gps_position->latitude_deg             = _buf.payload_rx_nav_pvt.lat * 1e-7;
> …
> PX4-Autopilot/src/drivers/gps/devices/src/emlid_reach.cpp:288:18: fatal error: no member named 'longitude_deg' in 'sensor_gps_s'
>                 _gps_position->longitude_deg = _erb_buff.payload.geodic_position.longitude;
> …
> PX4-Autopilot/src/drivers/gps/devices/src/nmea.cpp:269:18: fatal error: no member named 'longitude_deg' in 'sensor_gps_s'
>                 _gps_position->longitude_deg = int(lon * 0.01) + (lon * 0.01 - int(lon * 0.01)) * 100.0 / 60.0;
> …
> PX4-Autopilot/src/drivers/gps/devices/src/ashtech.cpp:244:18: fatal error: no member named 'latitude_deg' in 'sensor_gps_s'
>                 _gps_position->latitude_deg = static_cast<int>(lat * 0.01) + (lat * 0.01 - static_cast<int>(lat * 0.01)) * 100.0 / 60.0;
> …
> PX4-Autopilot/src/drivers/gps/devices/src/femtomes.cpp:101:18: fatal error: no member named 'latitude_deg' in 'sensor_gps_s'
>                 _gps_position->latitude_deg = _femto_uav_gps.lat / 1e7;
> …
> PX4-Autopilot/src/drivers/gps/devices/src/sbf.cpp:528:18: fatal error: no member named 'latitude_deg' in 'sensor_gps_s'
>                 _gps_position->latitude_deg = _buf.payload_pvt_geodetic.latitude * M_RAD_TO_DEG

It seems to be an issue of mismatch between the _**GPS library and the NuttX library**_. Has anyone else encountered this problem?

### To Reproduce

1. `git clone  https://github.com/PX4/PX4-Autopilot.git --recursive`
2. `make px4_sitl jmavsim`

### Expected behavior

I want it to compile successfully and run in the jmavsim simulator.

### Flight Log

`none`

### Software Version

I haven't reached that step yet.

### Flight controller

jmavsim simulator

### Vehicle type

None

bresch, 2023-08-18:
Did you install the while toolchain successfully? https://docs.px4.io/main/en/dev_setup/dev_env.html
If yes, have you tried to `git submodule sync --recursive && git submodule update --init --recursive`

mcsauder, 2023-08-18:
Also, reboot your computer after following all install instructions and updates.

swj735245824, 2023-08-18:
> Did you install the while toolchain successfully? https://docs.px4.io/main/en/dev_setup/dev_env.html If yes, have you tried to `git submodule sync --recursive && git submodule update --init --recursive`

Yes, I've done these. This is the result of executing '`git submodule sync --recursive && git submodule update --init --recursive`'：
[image]

After executing the previous command, I proceeded to execute the '`make px4_sitl jmavsim`' command, and here are the results：
[image]

The error shown in this image indicates the use of the '`latitude_deg`' property from the '`sensor_gps_s`' structure in the '`mtk.cpp`' file. However, this property does not exist within this structure. This issue is also reported not only in '`mtk.cpp:220:18`' but also in '`ashtech.cpp:244:18`' and '`ubx.cpp:1898:19`'.

swj735245824, 2023-08-18:
> Also, reboot your computer after following all install instructions and updates.

Yes, I followed the steps in the instructional video. I'm using `MacOS Big Sur 11.4` on my computer, and after installing the toolchain with `brew,` I've restarted a few times.

swj735245824, 2023-08-19:
> Did you install the while toolchain successfully? https://docs.px4.io/main/en/dev_setup/dev_env.html If yes, have you tried to `git submodule sync --recursive && git submodule update --init --recursive`

Hey, buddy, are you busy? Can I help you?

bresch, 2023-08-22:
output of `git status` ?
Looks like the submodules are not synced with the version of your code. 
You could also try `make distclean`

swj735245824, 2023-08-23:
> 

Thanks a lot. Tomorrow I will try the whole Ubuntu 20.04 LTS system on my desktop computer again. I feel like there's an issue with macOS's Git – encountered quite a few problems while cloning. I had to manually add the submodules.

mcsauder, 2023-08-23:
Maybe try Ubuntu's latest LTS distro, 22.04? Maybe update macOS as well?

bresch, 2023-08-23:
Closing for now as this was most likely an issue with your setup and not a PX4 issue, feel free to re-open it if needed.

closed 2023-08-23 by bresch

romain-chiap, 2024-03-09:
Same problem here
```/PX4-Autopilot/src/drivers/gps/devices/src/emlid_reach.cpp: In member function 'int GPSDriverEmlidReach::handleErbSentence()':
/PX4-Autopilot/src/drivers/gps/devices/src/emlid_reach.cpp:288:18: error: 'struct sensor_gps_s' has no member named 'longitude_deg'
  288 |   _gps_position->longitude_deg = _erb_buff.payload.geodic_position.longitude;
      |                  ^~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
...
/PX4-Autopilot/src/drivers/gps/gps.cpp: In member function 'virtual void GPS::run()':
/PX4-Autopilot/src/drivers/gps/gps.cpp:839:86: error: no matching function for call to 'GPSDriverUBX::GPSDriverUBX(GPSHelper::Interface&, int (*)(GPSCallbackType, void*, int, void*), GPS*, sensor_gps_s*, satellite_info_s*&, int32_t&, float&, GPSDriverUBX::UBXMode&)'
  839 |                                            gps_ubx_dynmodel, heading_offset, ubx_mod);
      |                                                                                     ^
compilation terminated due to -Wfatal-errors.
```

It seems like the .h files are ignored from the compilation. I'm on MacOS 10.15.7, and no I'm not going to update it as I want to keep compiling older versions of PX4. 

Any idea?