PX4/PX4-Autopilot #21983: PX4 connect AirSim through WSL2

AkumaET, 2023-08-19:
### Describe the bug

[  0%] Built target events_header
[  0%] Built target parameters_xml
[  0%] Built target ver_gen
[  0%] Built target uorb_headers
[  0%] Built target flighttasks_generated
[  0%] Built target actuators_json
[  1%] Built target output_functions_header
[  1%] Built target drivers_board
[  1%] Built target git_mavlink_v2
[  1%] Built target git_gps_devices
[  1%] Built target git_micro_xrce_dds_client
[  1%] Built target mode_util
[  1%] Built target uorb_ucdr_headers
[  2%] Built target topic_bridge_files
[  2%] Built target events_json
[  2%] Built target perf
[  3%] Built target geo
[  3%] Built target tinybson
[  3%] Built target version
[  3%] Built target wind_estimator
[  5%] Built target work_queue
[  5%] Built target world_magnetic_model
[  5%] Built target mavlink_c_generate_uavionix
[  6%] Built target libmicroxrceddsclient_project
[  6%] Built target component_general_json
[  6%] Built target mavlink_c_generate
[  6%] Built target parameters
[ 27%] Built target uorb_msgs
[ 28%] Built target romfs_gen_files_target
[ 28%] Built target px4_work_queue
[ 30%] Built target px4_platform
[ 30%] Built target lockstep_scheduler
[ 30%] Built target px4_daemon
[ 31%] Built target px4_layer
[ 31%] Built target airspeed
[ 31%] Built target adsb
[ 31%] Built target bezier
[ 31%] Built target hysteresis
[ 31%] Built target cdev
[ 31%] Built target battery
[ 31%] Built target circuit_breaker
[ 32%] Built target systemlib
[ 33%] Built target dataman_client
[ 33%] Built target conversion
[ 34%] Built target controllib
[ 34%] Built target CollisionPrevention
[ 34%] Built target l1
[ 34%] Built target mathlib
[ 34%] Built target motion_planning
[ 35%] Built target mixer_module
[ 35%] Built target npfg
[ 35%] Built target pid
[ 35%] Built target RateControl
[ 36%] Built target rc
[ 36%] Built target SlewRate
[ 36%] Built target sensor_calibration
[ 36%] Built target SystemIdentification
[ 36%] Built target tecs
[ 36%] Built target timesync
[ 36%] Built target tunes
[ 37%] Built target arch_tone_alarm
[ 37%] Built target WeatherVane
[ 37%] Built target failure_detector
[ 37%] Built target message_display
[ 37%] Built target failsafe
[ 37%] Built target ArmAuthorization
[ 42%] Built target health_and_arming_checks
[ 43%] Built target ActuatorEffectiveness
[ 43%] Built target ControlAllocation
[ 43%] Built target EKF2Utility
[ 43%] Built target follow_target_estimator
[ 44%] Built target FlightTask
[ 44%] Built target launchdetection
[ 44%] Built target runway_takeoff
[ 45%] Built target zero_order_hover_thrust_ekf
[ 46%] Built target AttitudeControl
[ 47%] Built target PositionControl
[ 48%] Built target Takeoff
[ 48%] Built target data_validator
[ 48%] Built target geofence_breach_avoidance
[ 48%] Built target vehicle_acceleration
[ 48%] Built target vehicle_imu
[ 48%] Built target vehicle_gps_position
[ 48%] Built target vehicle_angular_velocity
[ 48%] Built target vehicle_magnetometer
[ 49%] Built target vehicle_optical_flow
[ 49%] Built target avoidance
[ 50%] Built target uORB
[ 50%] Built target drivers_accelerometer
[ 50%] Built target drivers_gyroscope
[ 51%] Built target drivers_magnetometer
[ 51%] Built target FlightTaskFailsafe
[ 51%] Built target FlightTaskUtility
[ 52%] Built target modules__fw_att_control
[ 52%] Built target fw_autotune_attitude_control
[ 52%] Built target modules__fw_pos_control
[ 52%] Built target modules__fw_rate_control
[ 53%] Built target modules__gyro_calibration
[ 54%] Built target drivers__gimbal
[ 55%] Built target modules__gyro_fft
[ 56%] Built target modules__land_detector
[ 56%] Built target modules__load_mon
[ 56%] Built target modules__landing_target_estimator
[ 57%] Built target modules__local_position_estimator
[ 58%] Built target modules__manual_control
[ 60%] Built target modules__logger
[ 60%] Built target modules__mag_bias_estimator
[ 61%] Built target mc_autotune_attitude_control
[ 62%] Built target modules__mc_att_control
[ 62%] Built target modules__mavlink__mavlink_tests
[ 64%] Built target modules__mavlink
[ 64%] Built target modules__mc_pos_control
[ 64%] Built target modules__mc_hover_thrust_estimator
[ 64%] Built target modules__mc_rate_control
[ 64%] Built target modules__payload_deliverer
[ 64%] Built target modules__rc_update
[ 64%] Built target modules__rover_pos_control
[ 65%] Built target modules__replay
[ 66%] Built target vehicle_air_data
[ 66%] Built target modules__simulation__battery_simulator
[ 66%] Built target modules__simulation__pwm_out_sim
[ 66%] Built target modules__simulation__sensor_airspeed_sim
[ 67%] Built target modules__simulation__gz_bridge
[ 68%] Built target modules__simulation__sensor_baro_sim
[ 68%] Built target modules__simulation__sensor_gps_sim
[ 68%] Built target modules__simulation__senosr_mag_sim
[ 68%] Built target modules__simulation__simulator_sih
[ 68%] Built target modules__simulation__simulator_mavlink
[ 69%] Built target modules__temperature_compensation
[ 69%] Built target modules__uuv_att_control
[ 69%] Built target modules__uxrce_dds_client
[ 70%] Built target modules__uuv_pos_control
[ 71%] Built target modules__vtol_att_control
[ 72%] Built target systemcmds__actuator_test
[ 72%] Built target systemcmds__bsondump
[ 72%] Built target systemcmds__dyn
[ 72%] Built target systemcmds__failure
[ 72%] Built target systemcmds__led_control
[ 73%] Built target systemcmds__param
[ 73%] Built target systemcmds__perf
[ 73%] Built target systemcmds__sd_bench
[ 73%] Built target systemcmds__shutdown
[ 73%] Built target systemcmds__tests__hrt_test
[ 73%] Built target systemcmds__system_time
[ 77%] Built target systemcmds__tests
[ 78%] Built target systemcmds__topic_listener
[ 78%] Built target systemcmds__tune_control
[ 78%] Built target systemcmds__uorb
[ 78%] Built target systemcmds__ver
[ 78%] Built target systemcmds__work_queue
[ 78%] Built target modules__fake_imu
[ 80%] Built target modules__fake_gps
[ 80%] Built target examples__fake_magnetometer
[ 80%] Built target examples__hello
[ 81%] Built target examples__px4_mavlink_debug
[ 81%] Built target examples__px4_simple_app
[ 81%] Built target examples__work_item
[ 82%] Built target lib__cdev__test__cdev_test
[ 82%] Built target lib__controllib__controllib_test
[ 82%] Built target lib__rc__rc_tests
[ 82%] Built target modules__uORB__uORB_tests
[ 83%] Built target lib__work_queue__test__wqueue_test
[ 84%] Built target drivers__camera_trigger
[ 84%] Built target drivers__osd__msp_osd
[ 86%] Built target drivers__gps
[ 86%] Built target drivers__tone_alarm
[ 86%] Built target modules__airship_att_control
[ 87%] Built target modules__airspeed_selector
[ 87%] Built target modules__attitude_estimator_q
[ 87%] Built target modules__camera_feedback
[ 87%] Built target modules__control_allocator
[ 87%] Built target modules__dataman
[ 89%] Built target modules__commander
[ 89%] Built target modules__events
[ 94%] Built target modules__ekf2
[ 94%] Built target FlightTaskAuto
[ 94%] Built target FlightTaskDescend
[ 94%] Built target mission_feasibility_checker
[ 94%] Built target FlightTaskManualAltitude
[ 94%] Built target FlightTaskTransition
[ 94%] Built target modules__sensors
[ 94%] Built target FlightTaskManualAltitudeSmoothVel
[ 94%] Built target FlightTaskManualPosition
[ 94%] Built target FlightTaskAutoFollowTarget
[ 96%] Built target modules__navigator
[ 97%] Built target FlightTaskManualPositionSmoothVel
[ 98%] Built target FlightTaskManualAcceleration
[ 98%] Built target FlightTaskOrbit
[ 98%] Built target modules__flight_mode_manager
[100%] Built target px4
[100%] launching px4 none_iris (SYS_AUTOSTART=10016)
INFO  [px4] PX4 server already running for instance 0
gmake[4]: *** [src/modules/simulation/simulator_mavlink/CMakeFiles/none_iris.dir/build.make:72: src/modules/simulation/simulator_mavlink/CMakeFiles/none_iris] Error 255
gmake[3]: *** [CMakeFiles/Makefile2:9617: src/modules/simulation/simulator_mavlink/CMakeFiles/none_iris.dir/all] Error 2
gmake[2]: *** [CMakeFiles/Makefile2:9624: src/modules/simulation/simulator_mavlink/CMakeFiles/none_iris.dir/rule] Error 2
gmake[1]: *** [Makefile:2542: none_iris] Error 2
make: *** [Makefile:227: px4_sitl_default] Error 2

### To Reproduce

nano ~/.bashrc (set the "export PX4_SIM_HOST_ADDR= . . . .")

the issue happens when running: "make px4_sitl_default none_iris"

### Expected behavior

The expected thing is that PX4 connects with the AirSim drone successfully.

### Screenshot / Media

[image]

### Flight Log

None

### Software Version

The recent version and cloned from: https://github.com/PX4/PX4-Autopilot.git
VS: 2022
AirSim: latest
Unreal engine: 4.27.2

### Flight controller

None

### Vehicle type

None

AkumaET, 2023-08-19:
Version of Ubuntu is 22.04

closed 2023-08-21 by AkumaET

GateraGael, 2024-03-06:
Before running: make px4_sitl_default none_iris
You will need to do the following:

1) check control_ip (in settings.json) from WSL2
    - run ip address show and see inet under eth0

2) check LocalHostIp (in settings.json) from command prompt, powershell, or WSL2
    - run ipconfig and see IPv4 Address under WSL (Hyper-V firewall)
    - from WSL2 - cat /etc/resolv.conf | grep nameserver | awk '{print $2}'