PX4/PX4-Autopilot #21999: Unable to use TELEM2 on CUAV V6X

TianmingLi, 2023-08-24:
### Describe the bug

Unable to use TELEM2 on CUAV V6X

### To Reproduce

1. start Mavlink instance on TELEM2 `mavlink start -d  /dev/ttyS4`
2. Connect TELEM2 to computer through CP2102 USB to UART Bridge
3. Open QGroundControl 
4. QGC cannot connect to flight controller

### Expected behavior

QGC  connect to flight controller

### Flight Log

None

### Software Version

v1.13.3

### Flight controller

CUAV V6X

### Vehicle type

None

### Additional context

TELEM1, TELEM3, UART4 works in same method

https://discuss.px4.io/t/unable-to-use-telem2-on-cuav-v6x/33947

TianmingLi, 2023-08-24:
I tried V5+ flight controller ,

1. set MAV_1_CONFIG to TELEM2, 
2. reboot
3. the SER_TEL2_BAUD show up
4. change SER_TEL2_BAUD  to 57600
5. TELEM2 can connect to QGC link TELEM1

For V6X
1. set MAV_1_CONFIG to TELEM2, 
2. reboot
3. the SER_TEL2_BAUD not show up

bogdannedelcu, 2023-11-29:
I confirm the same behaivor on the PX4 build 1.14, while on version 1.13.2 thing work ok.

I tried to update them manually using MAVLink Console in QGC, using param set/param save but no success.

The consequence is I cannot connect to companion computer using TELEM2 port. I attach a zip with both parameters extracted from 2 devices. Once i downgraded the second device to 1.13.2 it started to work.

[parameters.zip](https://github.com/PX4/PX4-Autopilot/files/13498371/parameters.zip)

Mahmood-Hussain, 2025-06-20:
Any updates on this ? I faced the same issue with v1.15.4