PX4/PX4-Autopilot #22001: [Bug] The default.cmake file does not exist in board/px4/fmu v5 of version 1.13.3 of PX4.

Byzeal, 2023-08-24:
### Describe the bug

Hello, I am using version 1.13.3 of PX4 to learn some examples in src/example. I want to modify the examples and upload them to the hardware for verification and modification results, but I cannot find the default.cmake file in board/px4/fmu v5. I would like to inquire, does the default.cmake file no longer exist in version 1.13.3 of PX4? Where should I go to modify it?

### To Reproduce

I want to modify the examples and upload them to the hardware for verification and modification results,

### Expected behavior

The default.cmake file does not exist in board/px4/fmu v5

### Flight Log

None

### Software Version

v1.13.3

### Flight controller

HIL Quadcopter X

### Vehicle type

Multicopter

dagar, 2023-08-24:
@Byzeal due to growing complexity we migrated the board variant configuration to kconfig. The equivalent file is now `px4/fmu-v5x/default.px4board` instead of default.cmake. https://github.com/PX4/PX4-Autopilot/blob/410206aa5df6069c3f693c7e59db30091b3d170a/boards/px4/fmu-v5x/default.px4board

[image]

closed 2023-08-24 by dagar

Byzeal, 2023-08-26:
Thank you very much for your reply.