PX4/PX4-Autopilot #22002: [Bug] px4 1.14 rc1 down lidar error

youngjae41, 2023-08-25:
### Describe the bug

I'm using a down lidar because I'm flying indoors. Both baro and gps fusion are off and lidar setting is also set to 2: Enabled.
hgt is also set to 2: Range sensor, and the lidar range is 0.1cm to 10m.

I am using tfminipro lidar. 

At first, I tried to fly indoors using lidar and baro fusion, but the aircraft did not work properly due to the baro. So I changed all settings to only rangefinder

### To Reproduce

Make sure the range finder is working for it to work.

After that, they start take off and try to land very slowly from 1m to 0.5cm.

### Expected behavior

Between 0.8 and 0.1, the altitude display of qground changes to ---- <- momentarily

Then, after changing to --, the aircraft rises explosively upwards. When the altitude is reset, it will stop ascending and stop.

### Flight Log

i'm sorry 

i'm using px4 kakute h7 mini but px4 kakute h7 mini does not support saving logs.

### Software Version

 px4 1.14 rc1

### Flight controller

kakute h7 mini

### Vehicle type

None

ryanjAA, 2023-08-31:
@youngjae41 typically, and this is manufacturer specific, a -- of lidar data can mean no data but I would expect that to happen immediately after boot up, as in if data is not present on the bus (if i2c), maybe if serial then it could come and go and I would look for a loose wire but usually you get the out of bounds reading (a very high or low number).

Please post the log.

youngjae41, 2023-08-31:
@ryanjAA I also want to feed logs here. Is there a way to save logs in kakuteh7v2?

julianoes, 2023-10-19:
Closing this because I don't know what could be done. You would have to use a flight controller which allows logging to reproduce the issue.

closed 2023-10-19 by julianoes