PX4/PX4-Autopilot #22004: [Bug] Pixhawk 6x and 6x mini: 90 degrees offset after every compass calibration

MehdiBelal, 2023-08-25:
### Describe the bug

The most recent versions of v1.14 (including RC1, beta2, the latest/1.14, and main branches) are giving rise to problems concerning the px4_fmu_v6x target. Following compass calibration, there is a consistent issue wherein the yaw angle value consistently deviates by 90 degrees from the true value. The only remedy currently feasible is the deactivation of the MAG0 magnetometer via QGroundControl, and manual adjustment of the MAG1 orientation, which collectively re-establish a dependable yaw angle reading.

This behavior cannot be replicated in the case of px4_fmu_v5x on Pixhawk 5x devices, nor on any device equipped with version 1.13.3.

### To Reproduce

- Deploy px4_fmu_v6x target built from RC1, beta2, the latest/1.14, or main
- Reset all parameters to default 
- Sensors calibration (compass)
- Reboot 
- Yaw systematically deviated by 90 degrees

### Expected behavior

- Yaw aligned with correct orientation 

### Screenshot / Media

In Selection_001 the vehicle is pointing north, but the yaw estimation is deviated to east:
[image]

Mag1 sensor configuration after calibration:
[image]

Mag0 sensor configuration after calibration:
[image]

### Flight Log

Sensor calibration log:
https://logs.px4.io/plot_app?log=6e5d05cb-dbdc-422c-b085-c8b3c54bff00

Post calibration log:
https://review.px4.io/plot_app?log=c221d582-3097-4ee5-b055-b783c860c342

### Software Version

v1.14 RC1
v1.14 beta1, 
branch release/1.14 (21 Aug)
branch main branch (21 Aug)

### Flight controller

Pixhawk 6x, Pixhawk 6x mini

### Vehicle type

Multicopter

damien-robotsix, 2023-08-25:
After some digging from my side, it seems related to the definition of the rotation of the internal magnometer (at driver level) for the Pixhawk 6x

Forcing the rotation in rc.board_sensors allow to fix the issue, however, I don't fully understand how the different platforms are handled with the hwtypecmp variable, so I cannot propose a PR.

dagar, 2023-09-07:
As you've seen, during a successful mag calibration the external rotations are set to align with the internal magnetometer. 

Which hardware do you have exactly? I would suggest you start with opening a pull request with the change mentioned above and in the process I'll review it across all variants.

damien-robotsix, 2023-09-07:
It was seen on Holybro Pihawk6x (with and without the mini-set).

dagar, 2023-09-07:
To verify can you capture the output of `ver all`?

This board probably just needs to be included here? https://github.com/PX4/PX4-Autopilot/blob/35c1d5d9b1f80bb964f33a6686774d8a85415f77/boards/px4/fmu-v6x/init/rc.board_sensors#L105-L108

damien-robotsix, 2023-09-07:
> To verify can you capture the output of `ver all`?
> 
> This board probably just needs to be included here?
> 
> https://github.com/PX4/PX4-Autopilot/blob/35c1d5d9b1f80bb964f33a6686774d8a85415f77/boards/px4/fmu-v6x/init/rc.board_sensors#L105-L108

Maybe, I'll have a look, as told, I have no idea on how those tags are handled :D

damien-robotsix, 2023-09-07:
> To verify can you capture the output of `ver all`?
> 
> This board probably just needs to be included here?
> 
> https://github.com/PX4/PX4-Autopilot/blob/35c1d5d9b1f80bb964f33a6686774d8a85415f77/boards/px4/fmu-v6x/init/rc.board_sensors#L105-L108

`HW arch: PX4_FMU_V6X
HW type: V6X005003
HW version: 0x005
HW revision: 0x003
PX4 git-hash: 5359fe59e507896dace176d7fd8f9b0d00c2e3b6
PX4 version: 1.14.0 0 (17694720)
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: 3f77354c0dc88793a47ff3b57595195ab45f7ba9
Build datetime: Aug 25 2023 12:22:58
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 10.3.1 20210621 (release)
PX4GUID: 000600000000393137313132510100210027
MCU: STM32H7[4|5]xxx, rev. V`

dagar, 2023-09-07:
@damien-robotsix can you propose the change in a pull request and I'll track down anything else we need to verify for other hardware?

dagar, 2023-09-07:
There's still time to slip this in for v1.14.0

julianoes, 2023-09-08:
6X Mini:

```
ver all
HW arch: PX4_FMU_V6X
HW type: V6X005004
HW version: 0x005
HW revision: 0x004
PX4 git-hash: 3cf205c4a6d06073b1a0a1ac3caec3c03af926d7
PX4 version: 1.14.0 c0 (17694912)
PX4 git-branch: release/1.14
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: de41e7feaeffaec3ce65327e9569e8fdb553ca3d
Build datetime: Sep  8 2023 09:57:12
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 10.3.1 20210621 (release)
PX4GUID: 0006000000003235353833305105002c0029
MCU: STM32H7[4|5]xxx, rev. V
```

6X normal:

```
HW arch: PX4_FMU_V6X
HW type: V6X000004
HW version: 0x000
HW revision: 0x004
PX4 git-hash: 3cf205c4a6d06073b1a0a1ac3caec3c03af926d7
PX4 version: 1.14.0 c0 (17694912)
PX4 git-branch: release/1.14
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: de41e7feaeffaec3ce65327e9569e8fdb553ca3d
Build datetime: Sep  8 2023 09:57:12
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 10.3.1 20210621 (release)
PX4GUID: 00060000000037343637333051190042003d
MCU: STM32H7[4|5]xxx, rev. V
```

Both, the 6X and 6X mini seem to be correct with rotation 0 instead of 6. Which would match v1.13 where there is no rotation set:

https://github.com/PX4/PX4-Autopilot/blob/release/1.13/boards/px4/fmu-v6x/init/rc.board_sensors#L72-L78

It looks like this slipped in here:
https://github.com/PX4/PX4-Autopilot/commit/ed0f602ce8d8c75613c0a07f2de38c3de3f521c0#r97318847

And got merged like this:
https://github.com/PX4/PX4-Autopilot/pull/20721/files#diff-70d0f8c00e5bed2400f3d0f0e59f82806a8097b42766d3511d28918e71b90fc8R93

And then fixed for Skynode but not the rest:
https://github.com/PX4/PX4-Autopilot/pull/21194/files

So what's funny and confusing is that the rotation was reset to 0 for the Skynode models which according to @baumanta actually required rotation 6.

I will revert this to 0 (no rotation) for all, like it was with v1.13, and it's up to @baumanta and @marcoraeth to check Skynode one more time.

baumanta, 2023-09-08:
Thanks for fixing this @julianoes ! We had quite a bit of internal mess with that mag orientation, but I think it is correct now. Sorry for the confusion...

marcoraeth, 2023-09-08:
Thanks @julianoes. I also checked it and it is now correct. Rotation 6 was never correct

julianoes, 2023-09-11:
Thanks for checking everyone!

PR #22060 (2023-09-11): 6x: fix internal mag rotation
From looking at the history the BMM150 rotation was initially 0. Then, this was changed to 6 when the intent was to only change it for Skynode.

A bit later, the rotation was changed back to 0, but only for Skynode.

This tells me that rotation 0 was correct for all 6X including Skynode all along.

Tested on my Holybro 6X and 6X mini.

Fixes #22004.

Needs backporting to v1.14.

FYI @baumanta and @marcoraeth.
closed 2023-09-11 by julianoes