PX4/PX4-Autopilot #22016: [Bug] PX4 seems don't fuse external_vision position(x,y,z) and yaw

Wu-ZW, 2023-08-29:
### Describe the bug

After setting paraeters, using LIO( x y z yaw.) as external_vision informations.

### To Reproduce

1.Run LIO algorithm,publish "/mavros/vision_pose/pose" topic to px4
2.Take off
3.drone suddenly drifts away

### Expected behavior

The fused position is close to external_vision(LIO,Lidar-Interial-Odometry)

### Screenshot / Media

[image]

### Flight Log

https://logs.px4.io/plot_app?log=f1c325c3-2c38-4b82-85e5-a13aed3439b3

### Software Version

v1.13.2 (46a12a09)

### Flight controller

PX4_FMU_V6C

### Vehicle type

Multicopter

dagar, 2023-09-07:
Can you try to reproduce this with PX4 main or the v1.14.0 release candidate and I'll take a closer look at the log? I rewrote the EKF2 external vision handling between v1.13 and v1.14.

Make sure you set `SDLOG_PROFILE` to include "Estimator replay" and/or "Computer Vision and Avoidance" so that we can see the actual vehicle_visual_odometry data in your log.

julianoes, 2023-10-19:
No response in a while, closing.

closed 2023-10-19 by julianoes

Wu-ZW, 2023-11-01:
Thanks for your replay!!!! I will reproduce this with  v1.14.0 release candidate,it needs some time. I have read external vision part of EKF2, the reason why drifting is that "ev_data_ready" flag is false for a long time.