PX4/PX4-Autopilot #22026: Mission doesn't always start after arming (Fixed-wing)

ryanjAA, 2023-08-31:
### Describe the bug

As briefly discussed on the FW call with @tstastny. About 30-40% of the time if you arm the aircraft and immediately put it in mission mode from the RC remote, it doesn't move and throttle doesn't increase. It just stays armed and in mission mode. Usually a disarm then arm and back into mission mode fixes it (sometimes you have to do it twice but that is more rare). For good measure I like to re-upload the mission but that is not needed.

We did find COM_SPOOLUP_TIME but from the looks of the logs, it is greater than 1 second so this shouldn't apply (and maybe it has nothing to do with that param - never have changed it).

This has been seen countless times (>50 in recent months).

Here is a log that shows it happening but I only just brought it up and am hoping this log captures it (if not there are countless more I can dig up). 

More than anything it is annoyance but it should be something looked at.

I can't really get it to reproduce in SITL but that's likely because the params aren't exactly the same, etc.

https://review.px4.io/plot_app?log=8f8ad1f2-e26a-408b-872e-3407618f270d

### To Reproduce

Cant in SITL. Needs a code review most likely.

### Expected behavior

Immediately into mission mode and accelerating for runway takeoff

### Screenshot / Media

Can get the tlog if someone really wants to see...

### Flight Log

https://review.px4.io/plot_app?log=8f8ad1f2-e26a-408b-872e-3407618f270d

### Software Version

1.14 - specific commit in log

### Flight controller

6c

### Vehicle type

Fixed Wing

julianoes, 2023-10-19:
Still relevant?

ryanjAA, 2023-10-21:
Testing next week/week after. Will update this either way.

ryanjAA, 2023-11-03:
Still relevant @julianoes Flew today. I'll get the logs. I wonder if it is COM_SPOOLUP_TIME? on 1.14 RC2/Release

ryanjAA, 2023-11-03:
https://review.px4.io/plot_app?log=e986132b-db7b-49ee-8e74-58bf009a3132

ryanjAA, 2023-11-03:
You can see that it goes into mission and airspeed SP goes up but throttle does not increase.