PX4/PX4-Autopilot #22029: [Bug] ActuatorTest ORB_QUEUE_LENGTH value causes warning (non-pow-of-2)

spiderkeys, 2023-08-31:
### Describe the bug

The current ORB_QUEUE_LENGTH value of 12 in ActuatorTest.msg (as of 2985b5b9c28eb23c1d1ca3cb67de4ebcd7d2b539) causes warnings after more than one call to `actuator_test set`. 

Because the value is not a power of 2, on the first successful advertisement, it gets rounded up to 16. On the next call, the `_queue_size > queue_size` check fails (16 > 12).

```c++
int uORB::DeviceNode::update_queue_size(unsigned int queue_size)
{
	if (_queue_size == queue_size) {
		return PX4_OK;
	}

	//queue size is limited to 255 for the single reason that we use uint8 to store it
	if (_data || _queue_size > queue_size || queue_size > 255) {
		return PX4_ERROR;
	}

	_queue_size = round_pow_of_two_8(queue_size);
	return PX4_OK;
}
```
This doesn't seem to cause any harm other than triggering the following warning in `orb_advertise_multi`:
```c++
	/* Set the queue size. This must be done before the first publication; thus it fails if
	 * this is not the first advertiser.
	 */
	int result = px4_ioctl(fd, ORBIOCSETQUEUESIZE, (unsigned long)queue_size);

	if (result < 0 && queue_size > 1) {
		PX4_WARN("orb_advertise_multi: failed to set queue size");
	}
```
Should the queue size change to not trigger this behavior, or should the warning process change?

It's also worth noting that this occurs because the actuator_test systemcmd creates a new Publisher<actuator_test_s> every time the command is invoked, requiring the advertisement/queue size update. This wouldn't happen on a persistent Publisher.

### To Reproduce

1. Flash a board with the current main branch
2. Set the SYS_AUTOSTART to some vehicle (4001 quad, as an example), set SYS_AUTOCONFIG 1 and reboot
3. Run `actuator_test set -m 1 -v 0` twice

Warning will print on any subsequent run after the first:
```
nsh> 
nsh> actuator_test set -m 1 -v 0
WARN  [uORB] Update 'actuator_test' with qs [12] from [1]
INFO  [actuator_test] Press Enter to stop
nsh> actuator_test set -m 1 -v 0
WARN  [uORB] Update 'actuator_test' with queue size 12 from 16
WARN  [uORB] orb_advertise_multi: failed to set queue size
INFO  [actuator_test] Press Enter to stop
```

### Expected behavior

There shouldn't be a warning.

### Flight Log

Don't have one.

### Software Version

main: 3d61ab84c42ff8623bd48ff0ba74f9cf26bb402b

### Flight controller

H743-Wing

### Vehicle type

None

dagar, 2023-09-07:
Maybe I'm missing the problem here, but how about we...
 - make sure any in tree ORB_QUEUE_LENGTH is set appropriately (12 -> 16, etc)
 - adjust the check in `uORB::DeviceNode::update_queue_size()` so we round up first

> This wouldn't happen on a persistent Publisher.

Do you have something different in mind for this kind of systemcmd invocation?

spiderkeys, 2023-09-07:
> * make sure any in tree ORB_QUEUE_LENGTH is set appropriately (12 -> 16, etc)
> * adjust the check in `uORB::DeviceNode::update_queue_size()` so we round up first

Either or both sound good and would fix the warning. I'd been running with the same changes on my local branch, but wasn't sure if this was the desired solution.

> Do you have something different in mind for this kind of systemcmd invocation?

Not really, they are probably fine as is - the systemcmds don't currently have an object with a lifecycle that sticks around to persist any other objects or state (like publishers/subscribers), and it feels like introducing a singleton or static uORB pub/sub instances to solve that might potentially lead to headaches around construction/destruction ordering. I was just noting here that we only end up seeing this warning with systemcmds because they necessarily reconstruct the uorb publisher upon every invocation. Your two changes above will fix it regardless of this aspect.