PX4/PX4-Autopilot #22037: [Bug] drone is shaking after MC_PITCH_P value is changed

patric6123, 2023-09-02:
### Describe the bug

After MC_PITCH_P value is changed to a very high or low value, the drone is shaking.

### To Reproduce

1. run px4 gazebo
2. Take off
3. set MC_PITCH_P 21.5 ( Quadrotor ), 0, 345 ( Hexarotor ) or 0.01 ( VTOL Standard )

### Expected behavior

The drone flies normally.

### Screenshot / Media

[mc_pitch_p_21_5.webm](https://github.com/PX4/PX4-Autopilot/assets/143891125/1f57ceb5-2765-412d-9330-5fec46dc4fb5)

### Flight Log

.

### Software Version

1.12.3

### Flight controller

Quadrotor, Hexarotor, VTOL Standard

### Vehicle type

None

mcsauder, 2023-09-02:
Update this issue after upgrading to 1.14 if your issue persists. 1.12 was about 4 years ago.

dagar, 2023-09-07:
Duplicate of https://github.com/PX4/PX4-Autopilot/issues/22052

closed 2023-09-07 by dagar