PX4/PX4-Autopilot #22040: The fixedwing 's motor pwmoutput should not switch to the disarm value when the input of throttle at the lowest point

kevinuav, 2023-09-04:
### Describe problem solved by the proposed feature

The fixedwing 's motor pwmoutput will suddenly switch to disarm value rather than the minimum value when the input is near the lowest point in the latest firmware v1.14.0-main. which will cut off the gas engine while it is flying in the air. Could I configure it just outputs the disarm value only when it is disarmed? thank you.

### Describe your preferred solution

Can we make a parameter to configure it just outputs the disarm value only when it is disarmed? 

### Describe possible alternatives

The same as above.

sfuhrer, 2023-09-26:
There's a way you could configure it in all altitude controlled modes, but not yet when flying in Stabilized or Manual I think. I guess that's what you also need?
It currently is hard-coded to switch off at 2% (go to disarm), and I agree that we likely need to make this customizable. See https://github.com/PX4/PX4-Autopilot/blob/b21ad6af14dfc3c95c0d912748fcfaf23e7c7a93/src/modules/control_allocator/ActuatorEffectiveness/ActuatorEffectiveness.cpp#L94

kevinuav, 2023-09-26:
> There's a way you could configure it in all altitude controlled modes, but not yet when flying in Stabilized or Manual I think. I guess that's what you also need? It currently is hard-coded to [...]

Hi, Silvan. Thank you for your answer. But I don't know why the stabilized mode is different from the altitude controlled mode at this level since all the motor control values have been translated into the normalized values for effectiveness matrix?

sfuhrer, 2023-09-26:
You're correct about it being limited at the same place. But in all modes with altitude control you can specify the minimum thrust the controller outputs through FW_THR_MIN. If you set it above 0.02 then the motors will never go to disarmed. 
In stabilized you directly set the thrust from 0 to 100%.

kevinuav, 2023-09-28:
Hi, I have switch back to an early version firmware which has no this function: stopMaskedMotorsWithZeroThrust();
But it still has the same issue. Why? It seems like there something else makes the problem.

sfuhrer, 2023-09-29:
Which flight modes? Did you configure FW_THR_MIN?

kevinuav, 2023-09-30:
> Which flight modes? Did you configure FW_THR_MIN?

No,I didn't. It is the default value 0%.

ryanjAA, 2025-03-17:
@sfuhrer whats the rationale for this? Aka why not just delete the five lines that control this? Just trying to understand why it’s in there in the first place.

sfuhrer, 2025-03-17:
Because we want to make sure that the motor of a fixed-wing vehicle stops spinning completely when the throttle stick is at a minimum, and not spin at minimum PWM (as we do on multicopters).

ryanjAA, 2025-03-17:
Tell me if you do it differently but on all my setups, 0 throttle is 0. Aka, the motor doesnt spin. It is just the min PWM value to be at when armed but not actually moving anything. There's usually another 20-50 μs before we get going. Are you saying on your setup min pwm is actually slightly moving (if this wasn't in place)?

Because as far as the definition of things:

PWM_MAIN_MIN1 (INT32)
MAIN 1 Minimum Value.

Minimum output value (when not disarmed).

I dont think we say minimum output equating to movement. 

I just think that Min/Max should be a range (which it is). Whether you decide to have your min make things slightly move (aka then set it a little higher), or your max lower (maybe prop or battery constraints), that should be what you get. Not a hidden hardcode number.

At a minimum this should be a param.

It doesnt personally impact me but it feels wrong (and definitely will impact the ICE/turbine people.

sfuhrer, 2025-03-26:
>Are you saying on your setup min pwm is actually slightly moving (if this wasn't in place)?

For FW I do it the same as you, MIN is not spinning the motors. But for VTOLs MIN is set such that it still spins, and we need to have a way to tell the output driver to go below that in fixed-wing mode such that the motor can stop there.

The other consideration is that a only barely spinning motor creates drag and not thrust, which is something you can work around by setting the PWM_MIN to the value that actually causes a positive thrust, while still having a way to completely stopping the motor.

I agree that
- the 2% threshold is quite magic and should be considered to be parametrized. On fuel engine setups it would then be easy to disable it. 
- using PWM_DISARM if a motor is "Stopped" is not ideal, and should be documented at least