PX4/PX4-Autopilot #22103: [Bug] Processing of Euler312Yaw seems wrong

awanka, 2023-09-19:
### Describe the bug

https://github.com/PX4/PX4-Autopilot/blob/7ff00db9c59cd7a3d319bd66cf41420f0a9189a6/src/lib/mathlib/math/Utilities.hpp#L118

The function is unit tested by 
TEST(euler312YawTest, fromQuaternion)
https://github.com/PX4/PX4-Autopilot/blob/6d71224cdb1b14825fe65c5bf76b980662ec61a9/src/lib/mathlib/math/test/UtilitiesTest.cpp#L52C35-L52C35

However, the unit test calls getEuler321Yaw(), NOT getEuler312Yaw().
When changing to getEuler312Yaw(), the test fails.

This may point to a misunderstanding of Euler angles in the code.
Given the same rotation (fixed matrix or quaternion), the z axis rotation angle is different for Euler 3-2-1 vs. Euler 3-1-2. The code implicitly assumes that it is the same number by calling it yaw, see also updateEuler312YawInRotMat() and updateYawInRotMat().

### To Reproduce

Fix typo in 
https://github.com/PX4/PX4-Autopilot/blob/6d71224cdb1b14825fe65c5bf76b980662ec61a9/src/lib/mathlib/math/test/UtilitiesTest.cpp#L52C35-L52C35
to test getEuler312Yaw()
See unit test fail

### Expected behavior

Fixed unit test does not fail.

### Flight Log

n/a

### Software Version

v1.13.2

### Flight controller

n/a

### Vehicle type

None

bresch, 2023-09-20:
I think the name of the unit test is just wrong. The sequence used in `Matrix::Euler` is 3-2-1, so it makes sense that it matches the 321 sequence and not 312. To test 312 we should actually write another unit test

awanka, 2023-09-20:
Yes, fair enough, but what would the new unit test be?
Compare the output of getEuler321Yaw() vs. getEuler312Yaw()? This would fail, as the two 'yaw' values are not the same.
The underlying problem seems to be that they are treated as the same elsewhere in the code, e.g., in updateYawInRotMat().

bresch, 2023-09-21:
> The underlying problem seems to be that they are treated as the same elsewhere in the code, e.g., in updateYawInRotMat().

Well, yes and no. The thing is that for the estimator itself, switching from 321 to 312 isn't an issue because the measurement and the prediction are computed with the same sequence. What we then use to update the attitude estimate is the difference between the measurement and the prediction.

Nonetheless, I agree that having to switch between rotation sequences to avoid a gimbal lock isn't so nice. We're open for suggestions.

closed 2026-03-16 by dakejahl