PX4/PX4-Autopilot #22119: [Bug] ninja: error: unknown target 'gz_x500' when installing PX4 Simulator

idorobotics, 2023-09-21:
### Describe the bug

### Describe the bug

Following the latest [PX4 Gazebo Simulation instructions](https://docs.px4.io/main/en/sim_gazebo_gz/) installs Gazebo Garden but the PX4 simulator cannot be run. The error when running `make px4_sitl gz_x500` is `ninja: error: unknown target 'gz_x500'`. The error is the same for all available models. 

ROS 2 works fine on its own. 
Gazebo Garden also works fine on its own. 

Performing a make clean on PX4 Autopilot and running the install script and make does not solve the problem. 

### To Reproduce

1. Installed PX4 (main branch) and ROS 2 humble using the steps here: https://docs.px4.io/main/en/ros/ros2_comm.html#humble
2. Installed Gazebo Garden using the steps here: https://docs.px4.io/main/en/sim_gazebo_gz/
3. run `make px4_sitl gz_x500` and the output is `ninja: error: unknown target 'gz_x500'`
4. run `make --trace px4_sitl gz_x500` (output below)

```
Makefile:232: target 'px4_sitl' does not exist
# check if the desired cmake configuration matches the cache then CMAKE_CACHE_CHECK stays empty
# change to build folder which fails if it doesn't exist and CACHED_CMAKE_OPTIONS stays empty
# fetch all previously configured and cached options from the build folder and transform them into the OPTION=VALUE format without type (e.g. :BOOL)
# transform the options in CMAKE_ARGS into the OPTION=VALUE format without -D
# find each currently desired option in the already cached ones making sure the complete configured string value is the same
# if the complete list of desired options is found in the list of verified options we don't need to reconfigure and CMAKE_CACHE_CHECK stays empty
# make sure to start from scratch when switching from GNU Make to Ninja
if [ Ninja = "Ninja" ] && [ -e "~/PX4_software/PX4-Autopilot/build/px4_sitl_default"/Makefile ]; then rm -rf "~/PX4_software/PX4-Autopilot/build/px4_sitl_default"; fi
# make sure to start from scratch if ninja build file is missing
if [ Ninja = "Ninja" ] && [ ! -f "~/PX4_software/PX4-Autopilot/build/px4_sitl_default"/build.ninja ]; then rm -rf "~/PX4_software/PX4-Autopilot/build/px4_sitl_default"; fi
# only excplicitly configure the first build, if cache file already exists the makefile will rerun cmake automatically if necessary
if [ ! -e "~/PX4_software/PX4-Autopilot/build/px4_sitl_default"/CMakeCache.txt ] || [  ]; then mkdir -p "~/PX4_software/PX4-Autopilot/build/px4_sitl_default" && cd "~/PX4_software/PX4-Autopilot/build/px4_sitl_default" && cmake "~/PX4_software/PX4-Autopilot" -G"Ninja" -DCONFIG=px4_sitl_default || (rm -rf "~/PX4_software/PX4-Autopilot/build/px4_sitl_default"); fi
# run the build for the specified target
cmake --build "~/PX4_software/PX4-Autopilot/build/px4_sitl_default" --  gz_x500
ninja: error: unknown target 'gz_x500'
make: *** [Makefile:232: px4_sitl] Error 1
```

### Expected behavior

Run PX4 simulator in Gazebo Garden

### Flight Log

No flight report - cannot run simulator

### Software Version

Main branch from https://github.com/PX4/PX4-Autopilot

### Flight controller

No physical hardware connected.

### Vehicle type

None

dagar, 2023-09-21:
Can you give this a try?
 - first reboot if you haven't since installing gazebo
 - start from a clean state (make distclean)
 - run `make px4_sitl gz_x500` and share the full output

idorobotics, 2023-09-21:
Worked well, thanks! 
I think the issue was that following the [instructions to first set up ROS 2](https://docs.px4.io/main/en/ros/ros2_comm.html#humble) requires running `make px4_sitl` but the  [Gazebo setup instructions](https://docs.px4.io/main/en/sim_gazebo_gz/) expect you to run `make px4_sitl gz_x500` instead. Should I update the ROS2 User Guide docs to clarify?

closed 2023-09-22 by idorobotics

sanaosheng, 2024-08-07:
> Worked well, thanks! I think the issue was that following the [instructions to first set up ROS 2](https://docs.px4.io/main/en/ros/ros2_comm.html#humble) requires running `make px4_sitl` but the [...]

why I can run "make px4_sitl"seccessfully,but then failed to run "make px4_sitl gz_x500" anyway.

TZECHIN6, 2025-04-16:
Sorry for asking, as Gazebo official document suggest ubuntu22 to combine with ignition fortress. As i have previously working on a ground rover project using iginition already in the local machine. I have commented out the package of `gz-harmonic`. When I now trying to make the simulation of gz_x500, it fails to find the target...

Heres the log
```
thomasluk@inmsm-fractal-1:~/Code/PX4-Autopilot [main *=]$ make px4_sitl gz_x500
-- PX4_GIT_TAG: v1.16.0-beta1-104-g154623500e
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3") 
-- PX4 config file: /home/thomasluk/Code/PX4-Autopilot/boards/px4/sitl/default.px4board
-- PLATFORM posix
-- ROMFSROOT px4fmu_common
-- TESTING y
-- ETHERNET y
-- ROOT_PATH .
-- PARAM_FILE /fs/mtd_params
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- The CXX compiler identification is GNU 11.4.0
-- The C compiler identification is GNU 11.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- cmake build type: RelWithDebInfo
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Could NOT find gz-transport (missing: gz-transport_DIR)
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so (found version "3.12.4") 
-- Could NOT find gz-transport (missing: gz-transport_DIR)
-- Could NOT find gz-sim (missing: gz-sim_DIR)
-- Could NOT find gz-sensors (missing: gz-sensors_DIR)
-- Could NOT find gz-plugin (missing: gz-plugin_DIR)
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'bullet>=2.82'
--   Found bullet, version 3.05
-- Found fcl: /usr/include (found suitable version "0.7.0", minimum required is "0.5.0") 
-- Found assimp: /usr/include (found version "5.2.0") 
-- Found DART: /usr/include (Required is at least version "6.6") found components: dart 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time 
-- Looking for ignition-math6 -- found version 6.15.1
-- Searching for dependencies of ignition-math6
-- Looking for OGRE...
-- OGRE_PREFIX_WATCH changed.
-- Checking for module 'OGRE'
--   Found OGRE, version 1.9.0
-- Found Ogre Ghadamon (1.9.0)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Looking for ignition-math6 -- found version 6.15.1
-- Looking for ignition-transport8 -- found version 8.2.1
-- Searching for dependencies of ignition-transport8
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so (found suitable version "3.12.4", minimum required is "3") 
[... 28 lines cut ...]
-- Found JSONCPP: TRUE  
-- Checking for module 'libzip'
--   Found libzip, version 1.7.3
-- Found ZIP: TRUE  
-- Looking for ignition-common3 -- found version 3.14.0
-- Looking for ignition-math6 -- found version 6.15.1
-- Looking for ignition-msgs5 -- found version 5.8.1
-- Found gazebo-classic 11.14.0, including sitl_gazebo-classic simulator and gazebo-classic targets
CMake Warning at src/modules/simulation/simulator_mavlink/sitl_targets_gazebo-classic.cmake:149 (message):
  model directory
  /home/thomasluk/Code/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic/models/spacecraft_dart
  not found
Call Stack (most recent call first):
  src/modules/simulation/simulator_mavlink/CMakeLists.txt:57 (include)

-- Found Java: /usr/bin/java (found version "11.0.26") 
-- ROMFS: ROMFS/px4fmu_common
Architecture:  amd64
==> CPACK_INSTALL_PREFIX = @DEB_INSTALL_PREFIX@
-- Configuring done
-- Generating done
-- Build files have been written to: /home/thomasluk/Code/PX4-Autopilot/build/px4_sitl_default
ninja: error: unknown target 'gz_x500'
make: *** [Makefile:232: px4_sitl] Error 1
```

dakejahl, 2025-04-16:
@TZECHIN6 you can't have classic and gz installed simultaneously, for whatever reason. You need harmonic installed to build the gx_x500 target

TZECHIN6, 2025-04-16:
I can remove classic as its no longer used. however, can iginition still be used as its not gazebo classic in definition.

dakejahl, 2025-04-16:
@TZECHIN6 you need to use gazebo harmonic or later

Alansi775, 2025-04-24:
I am getting the same error is anyone solved this problem ninja: error: unknown target 'gz_x500'
make: *** [Makefile:232: px4_sitl] Error 1 and which Px4 version we should use and which ubuntu version?

dakejahl, 2025-04-24:
@Alansi775 try the main branch